lvgl/lvgl_esp32_drivers #5: Remove LV_DISPLAY_HEIGHT and LV_DISPLAY_WIDTH from display Kconfig

C47D, 2020-12-21:
That option is available in LVGL Kconfig.

VedantParanjape, 2020-12-30:
Hi, I encountered issue due to this. It is misleading to have two different parameters controlling the size of a display.

When will this be fixed ?

C47D, 2020-12-31:
I will be able to work on this over the weekend. First weekend of 2021!

VedantParanjape, 2021-01-04:
Hi, sounds good. If you need any help let me know. I mean assign this issue to me.

C47D, 2021-01-04:
Hi, I will look into this today, I built the project yesterday and had no issues. I will let you know later today if I assign this to you.

VedantParanjape, 2021-01-05:
> Hi, I will look into this today, I built the project yesterday and had no issues. I will let you know later today if I assign this to you.

Cool, it builds okay. but the docs completely missed this.

So, my code worked on 1st December and then I was trying to make it run on 31st December and it didn't work and I spent a good one day scratching my head, as to why it won't work.

C47D, 2021-01-06:
Can you try switching the lvgl_esp32_drivers submodule on lv_port_esp32 to follow [this ](https://github.com/lvgl/lvgl_esp32_drivers/tree/remove_redundand_config) branch?

VedantParanjape, 2021-01-07:
> Can you try switching the lvgl_esp32_drivers submodule on lv_port_esp32 to follow this branch?

I have been able to get it work for now, I'll wait till you merge into master

C47D, 2021-01-07:
Did you had any other issues? I haven't been able to test it with hardware, I only compiled the project.

VedantParanjape, 2021-01-07:
> Did you had any other issues? I haven't been able to test it with hardware, I only compiled the project.

no other issues, it worked with ssd1306 oled 128x64 module.

C47D, 2021-01-07:
Nice, I will merge the branch.

VedantParanjape, 2021-01-08:
Hi, @C47D. but in many places the code was using the macros that were removed.

C47D, 2021-01-08:
Lvgl configuration have both `LV_HOR_RES_MAX` and `LV_VER_RES_MAX`,  `LV_DISPLAY_ORIENTATION` was kept on the configuration options.

Are you getting compilation errors?

VedantParanjape, 2021-01-09:
@C47D https://github.com/lvgl/lvgl_esp32_drivers/blob/1cda575dac44c25be62df821abfff0756dcb7dd3/lvgl_helpers.h#L41

You are still using `CONFIG_LV_DISPLAY_WIDTH` and `CONFIG_LV_DISPLAY_HEIGHT` in many places in code, but you removed it from kconfig, so I guess it will throw a error, right ?

C47D, 2021-01-09:
You're right, I don't know why the compilation didn't failed, maybe the undefined symbols are set to 0.

VedantParanjape, 2021-01-10:
> You're right, I don't know why the compilation didn't failed, maybe the undefined symbols are set to 0.

It's going to be a nightmare to remove these. Just a suggestion, group all the macros from kconfig in one single header, easier to manage.

C47D, 2021-01-10:
I've already removed all references to those symbols, I forgot the name of the branch I've used. 

There's a lot of cleanup on the code base, the ultimate goal is to add all of the drivers into lv_drivers, for that we need to make them mcu/framework independent.

VedantParanjape, 2021-01-11:
> There's a lot of cleanup on the code base, the ultimate goal is to add all of the drivers into lv_drivers, for that we need to make them mcu/framework independent. 

You could look at HAL used by u8g2

VedantParanjape, 2021-01-11:
Don't know if this is relevant in this issue, let me know if you want me to move to new issue.

here's my code: https://pastebin.com/6EQgP0Dh

when I boot my board, it shows a static text "Text" on the upper left corner of the screen as follows

[image]

C47D, 2021-01-11:
@VedantParanjape let's move it into a new issue. Have you configured LV_SPRINTF_FLOAT_DISABLE? You're using a float format on your label.

VedantParanjape, 2021-01-11:
> @VedantParanjape let's move it into a new issue. Have you configured LV_SPRINTF_FLOAT_DISABLE? You're using a float format on your label.

Cool. I'll create a new issue in the main repository 👍🏻

> Also your for loop (right before the infinite loop) have no brackets. I will try to replicate your issue later today, got my stuff at home.

Ah, that was a typo, I'll fix it.

C47D, 2021-01-11:
Ok, let's follow over there.

closed 2021-01-15 by C47D