lvgl/lvgl_esp32_drivers #7: XPT2046 improvement discussion

liebman, 2020-12-24:
I've been having an issue with this input driver with spurious input events. This is where the IRQ line goes low for ~ 8us, measured with a scope.  Looking at other implementations I see that some compute the Z (or press pressure). Maybe this should only return a new input based on some Z threshold.

Now the datasheet is very confusing on this subject and gives formula that require knowing the X and or Y resistance values. 

Any thoughts?

liebman, 2020-12-24:
I've created a PR that works (at least for me).....

C47D, 2020-12-25:
Hi @liebman, I have also had problems with that touch sensor, for any reason the IRQ doesn't go low on my screen, so I also tried to implement the touch detection based on the Z coordinate.

liebman, 2020-12-25:
Hmm - this patch still requires the IRQ.  My issue was lots of spurious IRQ with no touch.  Should be simple enough to have the IRQ pin be optional, I may have time to try that this weekend.  Downside will be more traffic on SPI, in my case I'm sharing SPI with the display so I like the IRQ at least telling me I don't need to read the display.

liebman, 2020-12-25:
@C47D it was easier than I expected.  If you enable the `Check touch pressure to validate LV_TOUCH_PIN_IRQ` option for xp2046 then there is an additional option `Only use touch pressure and not IRQ to detect a touch` that will ignore IRQ (the pin will not even be configured) and only use pressure to determine a touch.  Let me know how this works for you.

PR #9 (2020-12-30): XPT2046 optionally (Kconfig option) validate a touch with pressure, not just IRQ low
Sometimes IRQ is very noisy an prevents things like idle time detection. This PR will check to insure there is pressure detected on the panel, not just the noisy IRQ signal. Disabled by default in Kconfig.
Fixes #7
closed 2020-12-30 by C47D