lvgl/lvgl_esp32_drivers #11: Q: Is there an advantage using this instead of TFT_eSPI

rvt, 2020-12-27:
I have an existing project based on Arduino and I do have some experience with TFT_eSPI so I know how this works so I wonder if there is any advantage of using esp32_drivers instead of TFT_eSPI.
I want to drive a ILI9341 display at 240x320.

Reason I am asking is because from what I currently understand is when using esp32_drivers/lv_port_esp32 I have to use RTOS instead of Arduino but I can be totally wrong here. I don´t fully understand yet why 
lv_port_esp32 exist..

But perhaps I should be looking at using lvgl_esp32_drivers and use the ILI9341 driver instead of TFT_eSPI?

C47D, 2020-12-28:
Hi @rvt,

This repo was initially created to show LVGL running in a ESP32 microcontroller (using an ILI9341 display), after some time more display controllers were added, as you already noticed this repo use the ESP-IDF framework from Espressif.

As far as I know the lvgl_esp32_drivers are coded to work with LVGL so they might be missing features available in TFT_eSPI.

rvt, 2020-12-28:
Hi @C47D I will play around with it and see what gives :) so far I am heaving fun coding the UI using SDL2 which works surprisingly well.

closed 2020-12-28 by rvt