lvgl/lvgl_esp32_drivers #13: Strange Adafruit TFT FeatherWing - 3.5" 480x320 issue

petestewardson-schneider, 2021-01-05:
Hi

I'm using the Adafruit 3.5"  TFT Featherwing with an Adafruit Huzzah32 board and having an issue.

The display appears to be working fine using this example code in that it it shows the write/list/chart example screens. However the touch operation is not working, and every 3s roughly the screen switches to the next of the three tabs in the example cycling around them. I am using the configuration for the TFT featherwing in menuconfig but this appears to also select settings for the touch controller without a way to change the CS so I'm not sure if it is this. 

I'm building on Windows with ESP-idf 4.0.1 and everything compiles and runs without error but not what I'd expect on screen.

C47D, 2021-01-05:
The error comes from here:

https://github.com/lvgl/lvgl_esp32_drivers/blob/6b1e219a8711bba4439e06b738662be4f6d92ca1/lvgl_touch/stmpe610.c#L74-L78

`0x811` is the `STMPE_CHIP_ID` reset value, maybe the spi configuration isn't properly done.

petestewardson-schneider, 2021-01-06:
@C47D right I tried removing SPI_DEVICE_HALFDUPLEX and it hasn't changed things. I've also tried a 10k pulldown to ground on MISO as I couldn't find a 4.7k. So far neither has made a difference. 

I do have a salae analyzer somewhere, I'll try to dig it out. 

The display is working though and they're both on the same SPI bus so the comms should be working, has to be config on the touch device I think

C47D, 2021-01-06:
I suggested removing the `SPI_DEVICE_HALFDUPLEX` flag because the communication is full duplex.

> I do have a salae analyzer somewhere, I'll try to dig it out.

Sure, thanks, please send the capture so I can take a look at it.

> The display is working though and they're both on the same SPI bus so the comms should be working, has to be config on the touch device I think

Sharing the SPI gave me problems before, but I haven't had time to check the code, my pc was very slow and I gave up, we might be missing something silly on the configuration.

michaelgaehwiler, 2022-07-20:
Hi,

I came also around this problem. I'm using Adafruit HUZZAH32 with Adafruit 3.5" Featherwing with ESP-IDF v4.4. With the Arduino component from Espressif it's working fully (Display, Touch, SD card). But not straight forward with lv_port_esp32 / lvgl_esp32_drivers. I would like to get rid of the Arduino component by switching to these projects.

Is it still possible to get help here?

Unfortunately, I don't have a logic analyzer. But I could give help by doing specific tests or having an interactive remote test session or even donate the hardware to somebody who would have the knowledge to deep-dive into problem solving.

Thanks a lot