lvgl/lvgl_esp32_drivers #18: Question: platformio support recommendation together for Arduino

rvt, 2021-01-12:
I am trying to use this using platformio with Arduino framework but I noticed that when I reference to this repository that all driver headers and c files are compiled. However, this is an issue when you just want to use one single driver (in my case ILI9341) because a lot of macro´s are not defined, specially a lot of error around EVE_XXX macro´s and what not.
I think this is because the header/c file will always be compiled by platformio where as in the cmake files they are excluded from the build based on menuconfig/cmake builds.

When I strip out all drivers except ILI9341 I can run the display fine, which is pretty cool I must say 👍 

Is there any trick/recommendation to use this from platformio builds?

liebman, 2021-01-12:
This is setup for `framework = espidf` and not `framework = arduino`.

C47D, 2021-01-12:
Pretty much what @liebman said, we are planning to move most of the drivers to the `lv_drivers` repo, but it will take a while.

I think people have been able to use the Platformio + Arduino+ LVGL + [TFT_eSPI](https://github.com/Bodmer/TFT_eSPI)

rvt, 2021-01-14:
I have used the TFT_eSPI but did not want to use that dependency, I only have some evidence from memory, but size of the firmware was a tad larger.

I believe your suggestion if using conditional compilation would work.
What could also work is providing defaults for all drivers so you don't get compile errors and let the linker decide what code is used and what not (I think  -Os takes care of that), I lack experience to understand if that's a good idea or not :)

For the time being, my stripped version of lvgl_esp32_drivers works good.

If you need any help testing let me know and I will keep a eye on https://github.com/lvgl/

closed 2021-01-14 by rvt