lvgl/lvgl_esp32_drivers #20: Display orientation doesn't change on ssd1306 on setting through menuconfig

VedantParanjape, 2021-01-15:
Display doesn't rotate on setting the option through menuconfig.

```c
void ssd1306_init()
{
	esp_err_t ret;
	i2c_cmd_handle_t cmd = i2c_cmd_link_create();

	i2c_master_start(cmd);
	i2c_master_write_byte(cmd, (OLED_I2C_ADDRESS << 1) | I2C_MASTER_WRITE, true);
	i2c_master_write_byte(cmd, OLED_CONTROL_BYTE_CMD_STREAM, true);

	i2c_master_write_byte(cmd, OLED_CMD_SET_CHARGE_PUMP, true);
	i2c_master_write_byte(cmd, 0x14, true);
    i2c_master_write_byte(cmd, 0xA0, true);
	i2c_master_write_byte(cmd, 0xC0, true);
	i2c_master_write_byte(cmd, OLED_CMD_SET_CONTRAST, true);
```

on changing init function as follows, the display rotates. Should I add the following feature through a PR ?

on line 150, in ssd1306.c, you have done this, there's no use of the macro, both if/else have the same code written, should this be removed ?
```c
#if defined CONFIG_LV_DISPLAY_ORIENTATION_LANDSCAPE		
    row1 = area->y1>>3;
    row2 = area->y2>>3;
#else
    row1 = area->y1>>3;
    row2 = area->y2>>3;
#endif
```

C47D, 2021-01-16:
I would rather use the `rotated` field on the `lv_disp_drv_t` pointer to know if the user wants to rotate the display, and yes, I'm working on improving the SSD1306 driver but haven't pushed anything to here.

VedantParanjape, 2021-01-16:
> I would rather use the `rotated` field on the `lv_disp_drv_t` pointer to know if the user wants to rotate the display, and yes, I'm working on improving the SSD1306 driver but haven't pushed anything to here.

rotated field rotates the display by 90 degree, I want 180 degree rotation. Can this be done using rotated field ?

C47D, 2021-01-17:
> rotated field rotates the display by 90 degree, I want 180 degree rotation. Can this be done using rotated field ?

I see, is this the code that you added to make it 180° rotated? It's at the `ssd1306_init` function.
```c
i2c_master_write_byte(cmd, 0x14, true);
i2c_master_write_byte(cmd, 0xA0, true);
i2c_master_write_byte(cmd, 0xC0, true);
```

> on line 150, in ssd1306.c, you have done this, there's no use of the macro, both if/else have the same code written, should this be removed ?

I was working on adding partial update support and removed that part of the code.

> Don't want to sound rude, but can you give a timeline for these changes, I want to use these in my project :) If necessary I can help you out, I have decent experience with esp-idf

I don't want to sound rude neither, but I only do this in my free time, and we have very different time zones lol.

C47D, 2021-01-17:
b7e1b51c7358f8a4738666a263a14a1cd3088131 should let you select portrait inverted orientation on the menuconfig and get the display rotated, let me know if it's works for you.

VedantParanjape, 2021-01-17:
> [b7e1b51](https://github.com/lvgl/lvgl_esp32_drivers/commit/b7e1b51c7358f8a4738666a263a14a1cd3088131) should let you select portrait inverted orientation on the menuconfig and get the display rotated, let me know if it's works for you.

Cool, I'll check this, we can actually rotate the screen on fly. Maybe we can have such a function, will be more flexible.

C47D, 2021-01-17:
I can try adding a function to rotate in 180°, in 90° would be different.

VedantParanjape, 2021-01-17:
> I can try adding a function to rotate in 180°, in 90° would be different. 

90° rotation for ssd1306 can be handled through software I guess. Cool.

C47D, 2021-01-19:
> 90° rotation for ssd1306 can be handled through software I guess. Cool.

Yes, I think so, we did something similar for the IL3820 driver.

Is there anything else we can do to close the issue?

VedantParanjape, 2021-01-20:
Oh, sorry. My issue is solved. I'll close it.

closed 2021-01-20 by VedantParanjape

michox, 2021-07-05:
the latest version seems to allow only Landscape orientation. What is the currently accepted solution for portrait mode? Do I need to rotate each element I want to render?

VedantParanjape, 2021-07-05:
> the latest version seems to allow only Landscape orientation. What is the currently accepted solution for portrait mode? I tried setting the disp_drv.rotated flag to 1 during initialization which will generate a portrait UI but it is still displayed in landscape orientation (the top half of the UI is displayed on the left half of the display, the bottom is not displayed and the right side of the display is empty)

Hi, you can set the rotation in kconfig.

C47D, 2021-07-05:
Hi @michox , adding support for portrait modes in the SSD1306 is a bit more difficult than other displays (e.g. ILI9341) because it doesn't seem to support rotation configuration, so we need to rotate it by software, please open a new issue to track this.

VedantParanjape, 2021-07-05:
@C47D Hi, you guys broke everything in v8.0, atleast should have maintained some backward compatibility.

I am using esp32, so should I already port to v8.0 or wait for esp32 component to update ?

C47D, 2021-07-05:
Hi, v8 is not "officially" supported in the repo, we're working on it.

VedantParanjape, 2021-07-05:
> Hi, v8 is not "officially" supported in the repo, we're working on it.

Okay, please update the readme, it directly asks to add lvgl master as submodule

michox, 2021-07-08:
> Hi, you can set the rotation in kconfig.

thanks for the reply. I can't seem to find the option though, could you please point me towards it? There are several kconfig files and I am new to LVGL. Thanks!

VedantParanjape, 2021-07-19:
> thanks for the reply. I can't seem to find the option though, could you please point me towards it? There are several kconfig files and I am new to LVGL. Thanks!

Hi, sorry for the late reply, I was wrong. You can't set portrait mode through menuconfig.

[image]