lvgl/lvgl_esp32_drivers #27: Question: Why is the default buffer size for the ILI9341 larger than for similar display controllers in lvgl_helpers.h?

jockm, 2021-02-04:
If we look in `lvgl_helpers.h` where `DISP_BUF_SIZE` is set/calculated you will see:

```
#if defined (CONFIG_LV_TFT_DISPLAY_CONTROLLER_ST7789)
#define DISP_BUF_SIZE  (LV_HOR_RES_MAX * 40)
#elif defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_ST7735S
#define DISP_BUF_SIZE  (LV_HOR_RES_MAX * 40)
#elif defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_ST7796S
#define DISP_BUF_SIZE  (LV_HOR_RES_MAX * 40)
#elif defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_HX8357
#define DISP_BUF_SIZE  (LV_HOR_RES_MAX * 40)
#elif defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_SH1107
#define DISP_BUF_SIZE  (LV_HOR_RES_MAX * LV_VER_RES_MAX)
#elif defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_ILI9481
#define DISP_BUF_SIZE  (LV_HOR_RES_MAX * 40)
#elif defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_ILI9486
#define DISP_BUF_SIZE  (LV_HOR_RES_MAX * 40)
#elif defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_ILI9488
#define DISP_BUF_SIZE  (LV_HOR_RES_MAX * 40)
#elif defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_ILI9341
#define DISP_BUF_SIZE  (LV_HOR_RES_MAX * 64)
```

Why is the `DISP_BUF_SIZE` 1.6x that of similar controllers?  I find it odd because it is very common for for ILI9341 based displays to be 320x240 and 64 doesn't divide cleanly into 240.  64 does however divide cleanly into 320, as in 480x320 displays we see associated the the ILI9486/8.

Also is there a reasoning behind the size of these buffers that could be documented?  If there were it would certainly help those who write their own drivers understand what their default buffer size should be

C47D, 2021-02-08:
@jockm Can you check if #30 works for you?

C47D, 2021-02-09:
Ok, I will merge it, thanks for the feedback!

commit 19a943c0 (2021-02-09): Merge pull request #30 from lvgl/fix/display_buffer_size

Enable using custom display buffer size, closes #27
closed 2021-02-09 by C47D