lvgl/lvgl_esp32_drivers #28: ESP32-S2 Projects cannot use pin numbers higher than 39 when configuring through menuconfig

jockm, 2021-02-04:
The ESP32-S2 has more available GPIOs than the ESP32, but menuconfig only allows assigning pins 0-39.

It remains clear that no testing or consideration of the -S2 chip has been given.

Again I must stress the need to publicly document that the -S2 chip is currently unsupported

C47D, 2021-02-04:
I've pushed a couple of commits that should allow you to select pins from 0 to 43 in the [fix/esp32-s2-gpios](https://github.com/lvgl/lvgl_esp32_drivers/tree/fix/esp32-s2-gpios)

[image]

Can you test it? You can do it by fetching the `lvgl_esp32_drivers` repo and then `git checkout --track origin/fix/esp32-s2-gpios'

jockm, 2021-02-04:
Works perfectly tested as both ESP32 and ESP32-S2 to see the pin ranges were appropriate for each, and looked at each pin assignments.  All good.  Ship It! ;)

C47D, 2021-02-04:
> All good. Ship It! ;)

Told you this issue was the easiest one.

PR #29 (2021-02-04): Fix/esp32 s2 gpios
Closes #28
closed 2021-02-04 by C47D