lvgl/lvgl_esp32_drivers #31: Unable to use multiple display types.

cmumford, 2021-02-08:
LVGL supports multiple displays, but lvgl_esp32_drivers will only compile one of the TFT display drivers, and there is no way to enable more than one display via menuconfig. CMakeLists.txt will only build one TFT driver ([see source](https://github.com/lvgl/lvgl_esp32_drivers/blob/9cd4a2d4c86729eed5237dd1cba6117debf9a368/lvgl_tft/CMakeLists.txt#L7)) and lvgl_helpers.c is coupled to the one enabled driver.

Can the configuration be modified to allow multiple enabled drivers? Another option might be to simply enable them all and rely on the linker to leave out the unused drivers.

C47D, 2021-02-08:
Hi,

I was working on that but had to solve some other issues. I was thinking on enabling the display controller related code and let the users do the configuration by themselves, as you already know the repo only works with one display in the project.

What displays do you need on your project?

cmumford, 2021-02-08:
My main display is a ILI9341, and supplemental is SSD1306.

C47D, 2021-02-09:
Ok, we are expecting some updates on the CMakeLists.txt file, so after that I can work on adding support for multiple displays on the project. I have not worked on a multiple-display project, so I would like to ask @kisvegabor and @embeddedt for tips on how can we improve this repo and `lv_port_esp32` to support that.