lvgl/lvgl_esp32_drivers #36: lvgl/lvgl.h: No such file or directory when using FT81X

Mair, 2021-02-11:
**Describe the issue**
When building the project, if I use a development board such as the WROVER kit v4.1 it builds correclty however, if I select to use a screen driver such as the FT81x I get a compilation error

**Code to reproduce the issue**
standard git clone (with recursion) on master

**Expected Results**
build correcltly

**Actual Results**
```
../components/lvgl_esp32_drivers/lvgl_tft/EVE_config.h:41:10: fatal error: lvgl/lvgl.h: No such file or directory
 #include "lvgl/lvgl.h"
          ^~~~~~~~~~~~~
compilation terminated.
```

**ESP32 Chip version**
N/A

**ESP-IDF version**
tried it on v4.1.1 as well as 4.0

**Development kit used**
N/A

**Development machine OS**
Tried on windows 10 as well as windows 8.1

**Compilation warnings/errors (if available)**
```
ir/FT81x.c.obj -MF esp-idf\lvgl_tft\CMakeFiles\__idf_lvgl_tft.dir\FT81x.c.obj.d -o esp-idf/lvgl_tft/CMakeFiles/__idf_lvgl_tft.dir/FT81x.c.obj   -c ../components/lvgl_esp32_drivers/lvgl_tft/FT81x.c
In file included from ../components/lvgl_esp32_drivers/lvgl_tft/EVE.h:39,
                 from ../components/lvgl_esp32_drivers/lvgl_tft/FT81x.c:8:
../components/lvgl_esp32_drivers/lvgl_tft/EVE_config.h:41:10: fatal error: lvgl/lvgl.h: No such file or directory
 #include "lvgl/lvgl.h"
          ^~~~~~~~~~~~~
compilation terminated.
[925/934] Building C object esp-idf/lvgl_tft/CMakeFiles/__idf_lvgl_tft.dir/disp_spi.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

**If possible, copy the compilation log into a file and attach it here**

C47D, 2021-02-11:
@tore-espressif Moved and thanks for the hint on were it broke!

Mair, 2021-02-11:
@tore-espressif and @C47D 
Thanks guys

coztemur, 2021-02-12:
@Mair , @tore-espressif , @C47D 
Thanks you very much for your support...

Mair, 2021-02-12:
@C47D Can you add a link to the issue / commit here?
Thanks

C47D, 2021-02-12:
@Mair a proposed fix was just merged into master, so you should be able to test it.

PR #37 (2021-02-12): Fix/simple include
Closes #36 

1. Fix include symbol in EVE_config.h
2. Fix symbol redefinition `TAG`. Previous definition here https://github.com/lvgl/lvgl_esp32_drivers/blob/master/lvgl_tft/EVE.h#L424
closed 2021-02-12 by C47D