lvgl/lvgl_esp32_drivers #48: SSD1306 menuconfig/kconfig broken

neg2led, 2021-03-13:
recent commit https://github.com/lvgl/lvgl_esp32_drivers/commit/9fed1cc47b5a45fec6bae08b55d2147d3b50260c conflicts somewhat with this block in kconfig:
https://github.com/lvgl/lvgl_esp32_drivers/blob/9fed1cc47b5a45fec6bae08b55d2147d3b50260c/lvgl_tft/Kconfig#L186-L206

kconfig does not actually _set_ `CONFIG_LV_DISPLAY_ORIENTATION_LANDSCAPE` or `CONFIG_LV_DISPLAY_ORIENTATION_LANDSCAPE_INVERTED`, it sets `CONFIG_DISPLAY_ORIENTATION_LANDSCAPE `and `CONFIG_DISPLAY_ORIENTATION_LANDSCAPE_INVERTED`

End result is that I get an invalid orientation error regardless of which orientation I set. I think this is just a simple kconfig update to fix, adding the LV prefix?

for now I've fixed it in my own stuff by picking the previous commit for the submodule in my project, but since [lvgl/lv_port_esp32](https://github.com/lvgl/lv_port_esp32) is using https://github.com/lvgl/lvgl_esp32_drivers/commit/9fed1cc47b5a45fec6bae08b55d2147d3b50260c as its referenced commit, it'll be broken for that repo/demo :(

C47D, 2021-03-14:
Hi,
Thanks for the report, I have been really busy with day work, but will fix it and update the demo project.

neg2led, 2021-03-19:
Hi, sorry, I thought I'd replied to this already... Yes, that's exactly what I did in my fork to fix it (revert of previous commit that changed those two lines). Works fine with that change 😄

PR #49 (2021-03-19): SSD1306: Fix orientation symbol names
Kconfig sets CONFIG_DISPLAY_ORIENTATION_LANDSCAPE and CONFIG_DISPLAY_ORIENTATION_LANDSCAPE_INVERTED.
This commit replaces the old references to CONFIG_LV_DISPLAY_ORIENTATION_LANDSCAPE and
CONFIG_LV_DISPLAY_ORIENTATION_LANDSCAPE_INVERTED.

Fixes: #48
commit 0442ebe7 (2021-03-19): SSD1306: Fix orientation symbol names

Kconfig sets CONFIG_DISPLAY_ORIENTATION_LANDSCAPE and CONFIG_DISPLAY_ORIENTATION_LANDSCAPE_INVERTED.
This commit replaces the old references to CONFIG_LV_DISPLAY_ORIENTATION_LANDSCAPE and
CONFIG_LV_DISPLAY_ORIENTATION_LANDSCAPE_INVERTED.

Fixes: #48
closed 2021-03-19 by C47D