lvgl/lvgl_esp32_drivers #50: Kconfig settings for TTGO T-Display faulty

AntonPill, 2021-03-24:
I discovered a problem when configuring a TTGO T-Display with menuconfig. when selecting the predefined Display setting.
First: the Items LV_TFT_DISPLAY_X_OFFSET and LV_TFT_DISPLAY_Y_OFFSET are not set correctly because of different item names DISPLAY_ORIENTATION_PORTRAIT vs. LV_DISPLAY_ORIENTATION_PORTRAIT and so on in  config LV_DISPLAY_ORIENTATION. After adding the "LV_" to the items in config LV_TFT_DISPLAY_X_OFFSET and config LV_TFT_DISPLAY_Y_OFFSET it works.

Second LV_TFT_DISPLAY_SPI_VSPI is not selected automatically because of the disabled prompt "TFT SPI Bus.". I made it visible by deleting the condition "!LV_PREDEFINED_DISPLAY_TTGO &&". This is only a workaround as i'm not confident with Kconfig i have no suggestion for a fix.

Onion2222, 2021-05-21:
Same problem

C47D, 2021-06-08:
Hi, I will try to fix this in the coming days, thanks for your patience.

jsmestad, 2021-06-26:
@C47D have you had a chance to look at this?

C47D, 2021-06-27:
Just merged your PR @jsmestad