lvgl/lvgl_esp32_drivers #51: Compilation errors in lvgl_helpers.c

ihouses, 2021-03-30:
I got some compilation errors in  lvgl_helpers.c

First:

#ifdef LV_LVGL_H_INCLUDE_SIMPLE
#include "src/core/lv_refr.h"      instead of  #include "src/**lv_core**/lv_refr.h"
#else
#include "lvgl/src/core/lv_refr.h"   instead of   #include "lvgl/src/**lv_core**/lv_refr.h"
#endif

The I get as well this:

../components/lvgl_esp32_drivers/lvgl_helpers.c:56:57: error: 'LV_HOR_RES_MAX' undeclared (first use in this function); did you mean 'LV_HOR_RES'?    
     ESP_LOGI(TAG, "Display hor size: %d, ver size: %d", LV_HOR_RES_MAX, LV_VER_RES_MAX);

Last issue looks like is caming from the past: https://github.com/lvgl/lv_port_esp32/pull/124

I am using the round screen driver GC9A01  resolution 240x240.

Can I change LV_HOR_RES_MAX and LV_VER_RES_MAX for 240?

C47D, 2021-06-08:
Working on it @embeddedt ... 🕐

C47D, 2021-06-08:
@ihouses Can you test the https://github.com/lvgl/lvgl_esp32_drivers/tree/feature/v8_support branch? I've added checks for the issues you've mentioned, but I don't have a setup to test them.

frezik, 2022-01-17:
I had the same problem, and using the features/v8_support branch got past that error. I'm now getting:

```
../components/lvgl_esp32_drivers/lvgl_helpers.c:22:10: fatal error: src/lv_core/lv_refr.h: No such file or directory
 #include "src/lv_core/lv_refr.h"
          ^~~~~~~~~~~~~~~~~~~~~~~
```