lvgl/lvgl_esp32_drivers #53: SSD1306 Wrong contrast initialisation and display mode

Alnef, 2021-04-04:
In the function `void ssd1306_init(void)` the configuration seems wrong:
```
    uint8_t display_mode = 0;

#if defined CONFIG_LV_INVERT_COLORS
    display_mode = OLED_CMD_DISPLAY_INVERTED;
#else
    display_mode = OLED_CMD_DISPLAY_NORMAL;
#endif

    uint8_t conf[] = {
        OLED_CONTROL_BYTE_CMD_STREAM,
        OLED_CMD_SET_CHARGE_PUMP,
        0x14,
        orientation_1,
        orientation_2,
        OLED_CMD_SET_CONTRAST,
        display_mode,
        0xFF,
        OLED_CMD_DISPLAY_ON
    };
```
We can see that the double byte command `OLED_CMD_SET_CONTRAST` (SSD1306 datasheet rev 1.1 p.28) is followed by `display_mode`.
So the contrast is set to 0xA6 or 0xA7 depending on the display mode configured and the display mode isn't configurable.

The fix that worked for me is to invert `display_mode` and `0xFF`:
```
    uint8_t conf[] = {
        OLED_CONTROL_BYTE_CMD_STREAM,
        OLED_CMD_SET_CHARGE_PUMP,
        0x14,
        orientation_1,
        orientation_2,
        OLED_CMD_SET_CONTRAST,
        0xFF,
        display_mode,
        OLED_CMD_DISPLAY_ON
    };
```

tore-espressif, 2021-04-09:
Can you please submit a Pull Request for this? Thanks!

C47D, 2021-05-10:
Sorry for the very late merge of the request @Alnef, life happened and I haven't checking the project lately. Feel free to close the issue if everything is working with your patch 😄

Alnef, 2021-05-22:
Thanks! It's working for me so I close the issue!

closed 2021-05-22 by Alnef