lvgl/lvgl_esp32_drivers #54: Touch on ESP32 Wroover and ILI9341 + XPT2046 not working.

Trackhe, 2021-04-09:
I use the espressif framework,
lvgl and 
lvgl_esp32_driver
and use it over HSPI

I have also loaded TFT_eSPI and here the touch is working. but after running 
    lv_init();
    lvgl_driver_init();

the init messages coming up in the monitor but the touch is not working.

Trackhe, 2021-04-27:
https://github.com/lvgl/lv_port_esp32/issues/273
I got the touch working. But the Ili9341 DP become grafik glitches with the benchmark. and the touch is working. the dp 3.5 zoll deisplays all correctly but the touch is not working. only x0 y0

tore-espressif, 2021-04-29:
@Trackhe  can we close this issue too or is there anything left to solve?

Trackhe, 2021-04-29:
the touch from the 3.5 zoll dp is not working. only the 2.8zoll dp. the touch controllers are the same. but the 3.5 zoll dp has the resolution 480x320 that was the reason why i had wrong values for the 2.8 zoll dp

Trackhe, 2021-04-29:
I guess i reopen the ticket if i have time to figure it out. Thank you for your nice Support. @tore-espressif

closed 2021-04-29 by Trackhe