lvgl/lvgl_esp32_drivers #56: Support for TK043F1569

ammaree, 2021-04-19:
### Introduce the problem

I recently purchased 2x displays with adapters to use with the ESP32 DevKit V4 (WROVER-IE) and the Espressif LCDKit v1.1
Based on advice from EspressIf (TK043F1508 ) and from users (link below) I ordered what I thought was the correct module.
[https://github.com/espressif/esp-iot-solution/issues/89#issue-674226075](url)

I purchased the display modules from the following link:
[https://buy.tbfocus.com/item.php?id=560872117142#3512775208514](url)

Upon receiving the modules I find that the pins are (slightly different) but more important that the controller has changed from NT35510 or RM68120 to LG4572

### Suggested solution
Any chance to get a driver adapted to support the LG4572 when used together with the DevKit and LCDKit mentioned above?

[image]

[Datasheet.hk_lg4572b_8093633.pdf](https://github.com/lvgl/lvgl/files/6325685/Datasheet.hk_lg4572b_8093633.pdf)

kisvegabor, 2021-04-19:
I've moved the issue to `lvgl_esp32_drivers` as this topic tightly related to ESP32.

ammaree, 2021-04-21:
Hi @kisvegabor 

Thanks for moving this, appreciated. Not being and LCD/LED/display technology expert I am unable to evaluate whether this display can work at all, or not. If you can you possibly advice on the following questions it will helps make a decision whether to abort the project and restart with different controllers.

Q1. Is it technically feasible to write a display driver to use this controller with the ESP32 DevKit V4 (WROVER-IE) and the Espressif LCDKit v1.1 ?

Q2. If possible, can you recommend the optimal starting point (existing driver) to use as foundation for developing an LG4572B driver.

Q3. If not, what makes the specific LG4572B controller incompatible with the  ESP32 DevKit V4 (WROVER-IE) and the Espressif LCDKit v1.1 ?

kisvegabor, 2021-04-22:
Hi,

I'm don't know this kit and the lg4572 controller. 

Looking into the datasheet you have linked the controller has SPI mode. See "5.5 SPI (Serial Peripheral Interface)".

However, from the photo of the module, it seems you can't select SPI mode, because only parallel modes are listed in the table 
(IM0..IM3). 
It might happen that the SPI mode works if you select it but it's not clear. :slightly_frowning_face: