lvgl/lvgl_esp32_drivers #60: Allow little-endian for ST7789

jhnlmn, 2021-05-06:
I found that I can insert
        {ST7789_DISPON, {0}, 0x80}, //existing line
#if LV_COLOR_16_SWAP == 0            // Test: enable little endian. 
        {ST7789_RAMCTRL, {0, 8}, 2}, // Test: enable little endian. 
#endif                               //DL:  enable little endian. 
        {0, {0}, 0xff}, //existing line
to st7789_init_cmds in st7789.c
and now I can build with CONFIG_LV_COLOR_16_SWAP=n
and make LVGL code 500 bytes smaller and more efficient.

Note that ST7789VW doc says "Note: Little Endian only can be supported in 65K 8-bit and 9-bit interface.", so this trick may not work for everyone, but it seem to work for me (I am using SPI, of course).

Also, it will be nice to remove "static" from "static void st7789_send_cmd(uint8_t cmd)" and "static void st7789_send_data(void * data, uint16_t length)"
so I can experiment with register programming from my own code.

Thank you

C47D, 2021-05-07:
Hi, I will try to add support for it and remove the static from the functions so you can add custom configuration, this weekend I might be free.

C47D, 2021-05-10:
I've pushed the branch `st7789_expose_cmd_api` where `st7789_send_cmd` and `st7789_send_data` are available outside `st7789.c`, can you test it?

jhnlmn, 2021-05-18:
Yes
I added
    #if LV_COLOR_16_SWAP==0
    //CONFIG_LV_COLOR_16_SWAP=n is more efficient for little endian platforms
    st7789_send_cmd(ST7789_RAMCTRL);
    st7789_send_data( (uint8_t[]) {0, 8}, 2);
    #endif
to guiTask in main.c
and it works OK for both 
CONFIG_LV_COLOR_16_SWAP=n
and
CONFIG_LV_COLOR_16_SWAP=y
I recommend CONFIG_LV_COLOR_16_SWAP=n
- code is smaller and more efficient.

Also, I suggest you add
#if LV_COLOR_16_SWAP == 0 
{ST7789_RAMCTRL, {0, 8}, 2}, 
#endif
to st7789_init