lvgl/lvgl_esp32_drivers #61: How do I tell lvgl that I only want to compile the driver for my ILI9488 board?

spierepf, 2021-05-18:
I've added `-D CONFIG_LV_TFT_DISPLAY_CONTROLLER_ILI9488` to my build flags, however, lvgl insists on compiling all the drivers regardless. This leads to errors in drivers that I'm just not interested in like:

```
.pio/libdeps/esp32doit-devkit-v1/lvgl_esp32_drivers/lvgl_tft/il3820.c:183:2: error: #error "Unsupported orientation used"
```

Is there a reason that chip-specific files don't honor the `CONFIG_LB_TFT_DISPLAY_CONTROLLER_*` flags?

Is there a better way to work around this issue besides renaming all the files I don't need?

tore-espressif, 2021-05-19:
Hello @spierepf ,
This should be handled in https://github.com/lvgl/lvgl_esp32_drivers/blob/master/CMakeLists.txt

Could you please share your project's `sdkconfig` file (located in root of your project)?

tore-espressif, 2021-05-19:
If you are using GNU Make, then there seems to be a bug:

Try deleting `lvgl_tft` on this line https://github.com/lvgl/lvgl_esp32_drivers/blob/master/component.mk#L4

spierepf, 2021-05-23:
Thanks @tore-espressif! I finally got it working. I needed to get the menuconfig option in platformio to work, and once I did I managed to make everything go. I posted my steps here: https://forum.lvgl.io/t/how-do-i-set-up-lv-platformio-with-lv-port-esp32/5542 in case others bump into a similar issue.

closed 2021-05-23 by spierepf