lvgl/lvgl_esp32_drivers #66: lv_disp_flush_ready() call missing from several flush callbacks

frhun, 2021-06-07:
Many if not most of the chipset drivers's flush callbacks are missing the `lv_disp_flush_ready()` call mandated by lvgl ( see [here](https://docs.lvgl.io/7.11/porting/display.html?highlight=lv_disp_drv_t#display-driver) ).

Is there any particular reason for this, or is it actually an oversight?

C47D, 2021-06-08:
Hi @frhun,

That call is being made in the SPI interrupt "callback", see https://github.com/lvgl/lvgl_esp32_drivers/blob/cccb932d3c8b0035632bfff866d242308d0ce167/lvgl_tft/disp_spi.c#L313

This interrupt callback is ESP32 specific.

frhun, 2021-06-08:
Ah, thanks. I missed that.

closed 2021-06-08 by frhun

morganrallen, 2021-09-20:
> That call is being made in the SPI interrupt "callback", see 

This might not be entirely true. With stock master branch I could not get the ST7796 to work until I added an explicit call as per #109. I cannot recall the exact step I took, but during my investigation I did note the `lv_disp_flush_ready` call in `disp_spi.c` but it only appeared to get called once then the driver would lock up, never returning from `lv_timer_handler`

C47D, 2021-09-20:
I will take a look at it.