lvgl/lvgl_esp32_drivers #68: Necessary updates for LVGL v8

C47D, 2021-06-08:
@tore-espressif @kisvegabor @embeddedt 
Is there anything special we should add to the drivers in order to support v8? Here's a list of what I think is necessary:

- [ ] Orientation awareness: Current rotation doesn't align with LVGL orientation, we have portrait, portrait inverted, landscape and landscape inverted, LVGL has orientation in 90° rotation.

kisvegabor, 2021-06-09:
There are minor changes on how the drivers are initialized. See [here](https://github.com/lvgl/lvgl/blob/master/docs/CHANGELOG.md#driver-changes).

C47D, 2021-06-13:
I think we also should update the rotate function on the drivers that has it, because the LVGL rotations are in 90° steps and it's different as we currently do it.

kisvegabor, 2021-06-14:
I agree.

icodk, 2021-06-17:
Updating to V8, I get the following Errors/Warnings: 
```
In file included from ../components/lvgl_esp32_drivers/lvgl_helpers.c:10:
../components/lvgl_esp32_drivers/lvgl_helpers.c: In function 'lvgl_driver_init':
../components/lvgl_esp32_drivers/lvgl_helpers.h:56:25: error: 'LV_HOR_RES_MAX' undeclared (first use in this function); did you mean 'LV_HOR_RES'?
 #define DISP_BUF_SIZE  (LV_HOR_RES_MAX * 40)
                         ^~~~~~~~~~~~~~
../components/lvgl_esp32_drivers/lvgl_spi_conf.h:135:34: note: in expansion of macro 'DISP_BUF_SIZE'
 #define SPI_BUS_MAX_TRANSFER_SZ (DISP_BUF_SIZE * 2)
                                  ^~~~~~~~~~~~~
../components/lvgl_esp32_drivers/lvgl_helpers.c:82:9: note: in expansion of macro 'SPI_BUS_MAX_TRANSFER_SZ'
         SPI_BUS_MAX_TRANSFER_SZ, 1,
         ^~~~~~~~~~~~~~~~~~~~~~~
../components/lvgl_esp32_drivers/lvgl_helpers.h:56:25: note: each undeclared identifier is reported only once for each function it appears in
 #define DISP_BUF_SIZE  (LV_HOR_RES_MAX * 40)
                         ^~~~~~~~~~~~~~
../components/lvgl_esp32_drivers/lvgl_spi_conf.h:135:34: note: in expansion of macro 'DISP_BUF_SIZE'
 #define SPI_BUS_MAX_TRANSFER_SZ (DISP_BUF_SIZE * 2)
                                  ^~~~~~~~~~~~~
../components/lvgl_esp32_drivers/lvgl_helpers.c:82:9: note: in expansion of macro 'SPI_BUS_MAX_TRANSFER_SZ'
         SPI_BUS_MAX_TRANSFER_SZ, 1,
         ^~~~~~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
```
Also the ver/hor parameters are missing from the V8 sdkconfig (Maximum horizontal/vertical to support by the library )
The roles of the lv_conf.h and sdkconfig are unclear. In the instructions for upgrading to V8 the lv_conf.h is mentioned but it is 
already not used in V7. In my opinion, all settings regarding lvgl should stay out of sdkconfig, which is an esp32 fenomen  and be set in lv_conf.h in the same way as for any other platform.

ropg, 2021-06-17:
Once I fixed those I got:

```c
../components/lvgl_esp32_drivers/lvgl_helpers.c:21:10: fatal error: src/lv_core/lv_refr.h: No such file or directory
 #include "src/lv_core/lv_refr.h"
```

I'd love to play with power management and sleep modes on v8 and am here if there's anything I can do to help make it work or to help document it. But people with more experience with LVGL and more of a feel for general direction probably need to take a first stab at adapting `lvgl_esp32_drivers`. And I can't play with power management in the simulator... I'll use the time to learn more about LVGL as a whole.

kisvegabor, 2021-06-17:
`lv_core` was renamed to `core`. (The `lv_` prefix is removed from the folder names)

C47D, 2021-06-17:
> ../components/lvgl_esp32_drivers/lvgl_helpers.c:21:10: fatal error: src/lv_core/lv_refr.h: No such file or directory
#include "src/lv_core/lv_refr.h"

This is solved in a PR on the `lvgl_esp32_drivers`  directory.

> Also the ver/hor parameters are missing from the V8 sdkconfig (Maximum horizontal/vertical to support by the library )
The roles of the lv_conf.h and sdkconfig are unclear. In the instructions for upgrading to V8 the lv_conf.h is mentioned but it is
already not used in V7. In my opinion, all settings regarding lvgl should stay out of sdkconfig, which is an esp32 fenomen and be set in lv_conf.h in the same way as for any other platform.

I think the `LV_HOR_RES_MAX` and `LV_VER_RES_MAX` should be in the display driver configuration on the menuconfig, but being able to configure the display drivers with the menuconfig or manually.

icodk, 2021-06-17:
I finally managed to compile an esp32  V8 but the lv_task_handler() never returns.
It manage to render part of a simple window object and dies. No reset, just nothing.
Same window (with v7 functions call) renders perfectly with V7: It get stack even I I don't call any function.
here is  the function I call just before entering the  lv_task_handler() loop:
```
void frmMain(void)
{
 lv_obj_t * win = lv_win_create(lv_scr_act(), 40);

	        lv_obj_t * btn;
	        btn = lv_win_add_btn(win, LV_SYMBOL_LIST, 40);
	        lv_obj_add_event_cb(btn, event_handler, LV_EVENT_CLICKED, NULL);

	        lv_win_add_title(win, "A title");

	        btn = lv_win_add_btn(win, LV_SYMBOL_BARS, 40);
	        lv_obj_add_event_cb(btn, event_handler, LV_EVENT_CLICKED, NULL);

	        btn = lv_win_add_btn(win, LV_SYMBOL_CLOSE, 60);
	        lv_obj_add_event_cb(btn, event_handler, LV_EVENT_CLICKED, NULL);

	        lv_obj_t * cont = lv_win_get_content(win);  /*Content can be aded here*/
	        lv_obj_t * label =  lv_label_create(cont);
	        lv_label_set_text(label, "This is\n"
	                                 "a pretty\n"
	                                 "long text\n"
	                                 "to see how\n"
	                                 "the window\n"
	                                 "becomes\n"
	                                 "scrollable.\n"
	                                 "\n"
	                                 "\n"
	                                 "Some more\n"
	                                 "text to be\n"
	                                 "sure it\n"
	                                 "overflows. :)");
}
```
In the above function, the title and the three buttons are drawn.

I also get the following compilation  warning : 
```
../main/main.c:151:23: warning: assignment to 'void (*)(struct _lv_indev_drv_t *, lv_indev_data_t *)' {aka 'void (*)(struct _lv_indev_drv_t *, struct <anonymous> *)'} from incompatible pointer type '_Bool (*)(lv_indev_drv_t *, lv_indev_data_t *)' {aka '_Bool (*)(struct _lv_indev_drv_t *, struct <anonymous> *)'} [-Wincompatible-pointer-types]
     indev_drv.read_cb = touch_driver_read;
```

embeddedt, 2021-06-17:
`touch_driver_read` needs to be updated to v8, otherwise, I'm pretty sure LVGL will just hang when it tries to read from the touchscreen. That update is being worked on in the `feature/esp32c3_support` branch.

icodk, 2021-06-17:
Updated both touch_drive.c and touch_drive.h to feature/esp32c3_support branch but still  lv_task_handler() does not returns
Using XPT2046. 
Any idea what I should do next ?
Thanks

icodk, 2021-06-18:
SOLVED !
First I got the the following error while the program still stuck at at the first call to lv_task_handler().
I got the following warning from the compiler:
```
../components/lvgl_esp32_drivers/lvgl_tft/disp_spi.c: In function 'spi_ready':
../components/lvgl_esp32_drivers/lvgl_tft/disp_spi.c:313:29: warning: passing argument 1 of 'lv_disp_flush_ready' from incompatible pointer type [-Wincompatible-pointer-types]
         lv_disp_flush_ready(&disp->driver);
                             ^~~~~~~~~~~~~
In file included from ../components/lvgl/src/hal/lv_hal.h:16,
                 from ../components/lvgl/lvgl.h:30,
                 from ../components/lvgl_esp32_drivers/lvgl_tft/disp_spi.c:23:
../components/lvgl/src/hal/lv_hal_disp.h:289:67: note: expected 'lv_disp_drv_t *' {aka 'struct _lv_disp_drv_t *'} but argument is of type 'struct _lv_disp_drv_t **'
 LV_ATTRIBUTE_FLUSH_READY void lv_disp_flush_ready(lv_disp_drv_t * disp_drv);
```
So I changed the  following line (file: disp_spi.c line 313) :
  lv_disp_flush_ready(&disp->driver);
to
  lv_disp_flush_ready(disp->driver);

and now everything works as expected and as in V7.
V8 looks great !

C47D, 2021-06-18:
I will try to replicate the issue, have you enabled logging?

icodk, 2021-06-19:
No logging.
 Which issue ? 
It is working for me now (as I wrote above)
Attached a complete working project (eclipse,idf-plugin,Windows 10) based on lvgl V8
[controlPanel.zip](https://github.com/lvgl/lvgl_esp32_drivers/files/6680781/controlPanel.zip)

C47D, 2021-06-21:
Sorry, I didn't saw you edited your reply, thanks for the attached project 😺

richardclli, 2022-02-08:
Is there a working branch for v8 support?

richardclli, 2022-02-24:
for LVGL v8, seems there are a lot of information are taken out from menuconfig and used the data type lv_disp_drv_t instead, screen size, orientation, etc. are all in there.  I think we may need to see how to modify the driver to cater for this change.