lvgl/lvgl_esp32_drivers #76: ST7735S: Display no longer works if RST pin is not selected

Rajssss, 2021-06-22:
Hello, thank you for maintaining this amazing library, saved a lot of my work.
I am trying to save some pins and decided not to use RST pin, hoping that the software reset will do the job, also I have connected RST pin to 3.3V.
But it just doesn't work. I have noticed that the source (st7735s.c) has the required code missing:

```
	//Initialize non-SPI GPIOs
        gpio_pad_select_gpio(ST7735S_DC);
	gpio_set_direction(ST7735S_DC, GPIO_MODE_OUTPUT);

#if ST7735S_USE_RST
        gpio_pad_select_gpio(ST7735S_RST);
	gpio_set_direction(ST7735S_RST, GPIO_MODE_OUTPUT);

	//Reset the display
	gpio_set_level(ST7735S_RST, 0);
	vTaskDelay(100 / portTICK_RATE_MS);
	gpio_set_level(ST7735S_RST, 1);
	vTaskDelay(100 / portTICK_RATE_MS);
#endif
```

I have tried adding `st7735s_send_cmd(TFT_SWRST);`
but still don't know why it doesn't work even after that, here is how it looks:
```
	//Initialize non-SPI GPIOs
        gpio_pad_select_gpio(ST7735S_DC);
	gpio_set_direction(ST7735S_DC, GPIO_MODE_OUTPUT);

#if ST7735S_USE_RST
        gpio_pad_select_gpio(ST7735S_RST);
	gpio_set_direction(ST7735S_RST, GPIO_MODE_OUTPUT);

	//Reset the display
	gpio_set_level(ST7735S_RST, 0);
	vTaskDelay(100 / portTICK_RATE_MS);
	gpio_set_level(ST7735S_RST, 1);
	vTaskDelay(100 / portTICK_RATE_MS);
#else
	st7735s_send_cmd(TFT_SWRST);
#endif
```

Not sure what else I should do to fix this issue.

tore-espressif, 2021-06-23:
[image]

Although, it should work with SW reset too.
Try adding 200ms delay before and after sending the SWreset command. I'll have a look if it doesn't help.

Rajssss, 2021-06-23:
> [image]
> 
> Although, it should work with SW reset too.
> Try adding 200ms delay before and after sending the SWreset command. I'll have a look if it doesn't help.

Didn't work. Same issue.

```
#ifdef CONFIG_LV_DISP_USE_RST
    gpio_pad_select_gpio(ST7735S_RST);
	gpio_set_direction(ST7735S_RST, GPIO_MODE_OUTPUT);

	//Reset the display
	gpio_set_level(ST7735S_RST, 0);
	vTaskDelay(100 / portTICK_RATE_MS);
	gpio_set_level(ST7735S_RST, 1);
	vTaskDelay(100 / portTICK_RATE_MS);
#else
	st7735s_send_cmd(ST7735_SWRESET);
	vTaskDelay(pdMS_TO_TICKS(200));
#endif
```

C47D, 2021-06-23:
I think I got one of this displays, will try to replicate the issue and make it work with software reset.

Rajssss, 2021-06-23:
> [image]
> 
> Although, it should work with SW reset too.
> Try adding 200ms delay before and after sending the SWreset command. I'll have a look if it doesn't help.

Didn't help.

```
#ifdef CONFIG_LV_DISP_USE_RST
    gpio_pad_select_gpio(ST7735S_RST);
	gpio_set_direction(ST7735S_RST, GPIO_MODE_OUTPUT);

	//Reset the display
	gpio_set_level(ST7735S_RST, 0);
	vTaskDelay(100 / portTICK_RATE_MS);
	gpio_set_level(ST7735S_RST, 1);
	vTaskDelay(100 / portTICK_RATE_MS);
#else
	st7735s_send_cmd(ST7735_SWRESET);
	vTaskDelay(pdMS_TO_TICKS(200));
#endif
	
```

Rajssss, 2021-06-23:
> [image]
> 
> Although, it should work with SW reset too.
> Try adding 200ms delay before and after sending the SWreset command. I'll have a look if it doesn't help.

Didn't help.

```
#ifdef CONFIG_LV_DISP_USE_RST
    gpio_pad_select_gpio(ST7735S_RST);
	gpio_set_direction(ST7735S_RST, GPIO_MODE_OUTPUT);

	//Reset the display
	gpio_set_level(ST7735S_RST, 0);
	vTaskDelay(100 / portTICK_RATE_MS);
	gpio_set_level(ST7735S_RST, 1);
	vTaskDelay(100 / portTICK_RATE_MS);
#else
	vTaskDelay(pdMS_TO_TICKS(200));
	st7735s_send_cmd(ST7735_SWRESET);
	vTaskDelay(pdMS_TO_TICKS(200));
#endif
```

Rajssss, 2021-06-25:
@C47D any update on this?

C47D, 2021-06-25:
No, sorry, a storm hit my city and just today we're getting back to normal. Will let you know any update over the weekend.

Rajssss, 2021-06-26:
@C47D @tore-espressif It seems to be working, but I had to connect the RST pin to the EN pin on my ESP32, and reduce the delay to 50ms. It didn't work if the RST pin is floating/VCC/GND. Also if I reset the esp32, sometimes the display works sometimes not, I don't know why.

C47D, 2021-06-26:
According to this image, the RST signal should go to Vcc (using a pull-up resistor)

[image]

Rajssss, 2021-07-01:
> According to this image, the RST signal should go to Vcc (using a pull-up resistor)
> 
> [image]

Don't know why but my display refused to work this way, I have to connect it to EN pin directly, but it works fine.
So It should be added in the code.

C47D, 2021-07-01:
Which EN pin do you mean?

Rajssss, 2021-07-02:
> Which EN pin do you mean?

The Enable pin on ESP32.