lvgl/lvgl_esp32_drivers #77: LVGL8 Not Working Maybe BUFFER Problems. ILI9341

Trackhe, 2021-06-24:
Device: ESP32 16mb flash
Display Res: 240x320

I tried some things. Custom Buffersize. 
set the DISP_BUF_SIZE like:

```
#if defined (CONFIG_CUSTOM_DISPLAY_BUFFER_SIZE)
    #define DISP_BUF_SIZE   CONFIG_CUSTOM_DISPLAY_BUFFER_BYTES
#elif (LVGL_VERSION_MAJOR < 8)
    #if defined (CONFIG_LV_TFT_DISPLAY_CONTROLLER_ST7789)
        #define DISP_BUF_SIZE  (LV_HOR_RES_MAX * 40)
    #elif defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_ST7735S
        #define DISP_BUF_SIZE  (LV_HOR_RES_MAX * 40)
    #elif defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_ST7796S
        #define DISP_BUF_SIZE  (LV_HOR_RES_MAX * 40)
    #elif defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_HX8357
        #define DISP_BUF_SIZE  (LV_HOR_RES_MAX * 40)
    #elif defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_SH1107
        #define DISP_BUF_SIZE  (LV_HOR_RES_MAX * LV_VER_RES_MAX)
    #elif defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_ILI9481
        #define DISP_BUF_SIZE  (LV_HOR_RES_MAX * 40)
    #elif defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_ILI9486
        #define DISP_BUF_SIZE  (LV_HOR_RES_MAX * 40)
    #elif defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_ILI9488
        #define DISP_BUF_SIZE  (LV_HOR_RES_MAX * 40)
    #elif defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_ILI9341
        #define DISP_BUF_SIZE  (LV_HOR_RES_MAX * 40)
    #elif defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_SSD1306
        #if defined (CONFIG_LV_THEME_MONO)
            #define DISP_BUF_SIZE  (LV_HOR_RES_MAX * (LV_VER_RES_MAX / 8))
        #else
            #define DISP_BUF_SIZE  (LV_HOR_RES_MAX * LV_VER_RES_MAX)
        #endif
    #elif defined (CONFIG_LV_TFT_DISPLAY_CONTROLLER_FT81X)
        #define DISP_BUF_LINES  40
        #define DISP_BUF_SIZE  (LV_HOR_RES_MAX * DISP_BUF_LINES)
    #elif defined (CONFIG_LV_TFT_DISPLAY_CONTROLLER_IL3820)
        #define DISP_BUF_SIZE (LV_VER_RES_MAX * IL3820_COLUMNS)
    #elif defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_RA8875
        #define DISP_BUF_SIZE  (LV_HOR_RES_MAX * 40)
    #elif defined (CONFIG_LV_TFT_DISPLAY_CONTROLLER_GC9A01)
        #define DISP_BUF_SIZE  (LV_HOR_RES_MAX * 40)
    #elif defined (CONFIG_LV_TFT_DISPLAY_CONTROLLER_JD79653A)
        #define DISP_BUF_SIZE ((LV_VER_RES_MAX * LV_VER_RES_MAX) / 8) // 5KB
    #elif defined (CONFIG_LV_TFT_DISPLAY_CONTROLLER_UC8151D)
        #define DISP_BUF_SIZE ((LV_VER_RES_MAX * LV_VER_RES_MAX) / 8) // 2888 bytes
    #elif defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_ILI9163C
        #define DISP_BUF_SIZE (LV_HOR_RES_MAX * 40)
    #endif
#else
    #if defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_ILI9341
        #define DISP_BUF_SIZE  (240 * 10)
    #else
        #error "No display controller selected or you dont set custom buffer and use the lvgl v8+ version"
#endif
#endif
```

i tried also 240 * 40
On the Display change a drawed white area but i cant see the label what i set. and the task that i use dosnt fire. 

LVGL programm part:
```
// Definition aller Objekte in dieser Klasse
lv_obj_t * gui_mainmenu_state;

// Tasks
static void gui_mainmenu_display_task_grad_update(lv_timer_t * timer);
static void gui_mainmenu_display_asyc_grad_update(void* p);
[... 736 lines cut ...]
I (5430) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
V bus_lock: dev 0 acquired.
V (5480) spi_master: polling trans
V (5480) spi_master: polling trans done
V (5480) bus_lock: dev 0 released.
V bus_lock: dev 0 acquired.
V (5480) spi_master: polling trans
V (5490) spi_master: polling trans done
V (5490) bus_lock: dev 0 released.
V bus_lock: dev 0 acquired.
V (5490) spi_master: polling trans
V (5500) spi_master: polling trans done
V (5500) bus_lock: dev 0 released.
V bus_lock: dev 0 acquired.
V (5510) spi_master: polling trans
V (5510) spi_master: polling trans done
V (5510) bus_lock: dev 0 released.
V bus_lock: dev 0 acquired.
V (5520) spi_master: polling trans
V (5530) spi_master: polling trans done
V (5530) bus_lock: dev 0 released.
V (5530) bus_lock: dev 0 served from bg.
```

Related: https://github.com/lvgl/lvgl_esp32_drivers/issues/68

C47D, 2021-06-24:
Hi,

We're just started porting both `lvgl_esp32_drivers` and `lv_port_esp32` to LVGL v8, I haven't been able to get my displays to test it at home, will try this weekend.

Trackhe, 2021-06-29:
> Hi,
> 
> 
> 
> We're just started porting both `lvgl_esp32_drivers` and `lv_port_esp32` to LVGL v8, I haven't been able to get my displays to test it at home, will try this weekend.

any news? i want to help :)

C47D, 2021-07-08:
Not really (from me), I'm a bit busy at work, but will try to take a look at this, I was trying to close all v7 related issues before starting to work on the v8 support.

Trackhe, 2021-07-08:
Understandable

ogghst, 2021-08-19:
i've found a possible fix. as per https://docs.lvgl.io/8.0/porting/display.html?highlight=lv_disp_draw_buf_t the flush callback must end with	lv_disp_flush_ready(drv). here is the modified function:

`void ili9341_flush(lv_disp_drv_t * drv, const lv_area_t * area, lv_color_t * color_map)
{
	uint8_t data[4];

	/*Column addresses*/
	ili9341_send_cmd(0x2A);
	data[0] = (area->x1 >> 8) & 0xFF;
	data[1] = area->x1 & 0xFF;
	data[2] = (area->x2 >> 8) & 0xFF;
	data[3] = area->x2 & 0xFF;
	ili9341_send_data(data, 4);

	/*Page addresses*/
	ili9341_send_cmd(0x2B);
	data[0] = (area->y1 >> 8) & 0xFF;
	data[1] = area->y1 & 0xFF;
	data[2] = (area->y2 >> 8) & 0xFF;
	data[3] = area->y2 & 0xFF;
	ili9341_send_data(data, 4);

	/*Memory write*/
	ili9341_send_cmd(0x2C);

	uint32_t size = lv_area_get_width(area) * lv_area_get_height(area);

	ili9341_send_color((void*)color_map, size * 2);

	lv_disp_flush_ready(drv);
}`

here is my fix, this at least makes the whole screen renderable, then I am working to speedup DMA buffer (now is limited to <4096 bytes to avoid overflow and is damn slow!)