lvgl/lvgl_esp32_drivers #79: Support for displays without DC pin?

Rajssss, 2021-06-27:
In menuconfig, there is a option to enable or disable the DC. However I beleive none of the drivers implemented it. Having such functionality will improve the usability of the driver, saving one more pin, and support Displays modules with no DC pin exposed.
Even though, I had hard time finding a driver which works without a DC pin, but found some hints to a 9-Bit SPI mode?

C47D, 2021-06-30:
Hi,

I think you have the ST7735S display, isn't? Can you test this branch? https://github.com/lvgl/lvgl_esp32_drivers/tree/feat/support_no_dc

Rajssss, 2021-06-30:
I will and let you know.

C47D, 2021-06-30:
The changes in that branch doesn't take into account the fact that the DC signal tells the display if the bytes it's getting from the MCU are a command or data.

Rajssss, 2021-07-01:
@C47D sorry it isn't working, getting garbage on the screen.

C47D, 2021-07-01:
I guess that's expected, the DC signals lets the display know if the data is a data or a command, what display are you working on? Could you share me an image of the connection?

Rajssss, 2021-07-02:
> I guess that's expected, the DC signals lets the display know if the data is a data or a command, what display are you working on? Could you share me an image of the connection?

Yes, but with DC pin selected it works fine as expected.
Here the display I have: http://www.lcdwiki.com/1.44inch_SPI_Module_ST7735S_SKU:MSP1443
And this is the pinout I am using.

[image]
[image]

C47D, 2021-07-02:
You just answered it, the display needs the DC signal to work, you can't use it without it.

Rajssss, 2021-07-02:
> You just answered it, the display needs the DC signal to work, you can't use it without it.

I don't get it. Are you saying that the display I have can not work without the DC pin? exactly what I am trying to achieve?
And is it because the DC pin is exposed in the module?

C47D, 2021-07-07:
> I don't get it. Are you saying that the display I have can not work without the DC pin?

Yes, I think so, your project works with the DC signal and it doesn't work without it.

> And is it because the DC pin is exposed in the module?

Yes.

The only displays I **think** work without DC signal are the I2C based ones.

Rajssss, 2021-07-07:
> Yes, I think so, your project works with the DC signal and it doesn't work without it.
>
>
> Yes.
>
> The only displays I **think** work without DC signal are the I2C based ones.

Unfortunately, I don't have one. maybe in future someone will test it.
I have found a thread on a display without DC pin exposed, using 9Bit SPI but for ILI9341, can you take a look?