lvgl/lvgl_esp32_drivers #81: Extra address-of operator in lv_disp_flush_ready(disp-driver)

cjheath, 2021-07-01:
https://github.com/lvgl/lvgl_esp32_drivers/blob/7d0bbe8ce57eece8abd955fceadc07c9bce7c3b4/lvgl_tft/disp_spi.c#L313

This line fails to compile for me, until I remove the ampersand.

C47D, 2021-07-01:
Hi @cjheath , I think somebody else reported the same problem in another issue, what version of LVGL are you using? is it v8 by any chance?

cjheath, 2021-07-01:
Yes, v8. Trying to follow the instructions for v7 not knowing there is a
partially complete transition was very difficult - several days wasted
chasing rabbits.

C47D, 2021-07-01:
Yes, we haven't started the transition to v8 just yet, trying to fix all the issues before that, we just merged the fix.

closed 2021-07-01 by C47D

cjheath, 2021-07-01:
Yeah that's all fine, but I just followed ESP-IDF instructions to check out trunk of lvgl and lvgl_esp32_drivers using git submodules - and assumed that what was on trunk would be CI tested to build cleanly. It's not a criticism of your development process (I'm cool with trunk-based development, though it's unfashionable), but if there is a better way for newbies to get a stable build, it should be easier to find. If I can create or improve the "getting started" instructions, please give me a pointer so I can help others avoid the pain I went through.
It's also weird that it still seems necessary to edit lv_conf.h to set LV_HOR_RES_MAX/LV_VER_RES_MAX. It's not a very Kconfig thing to do. Is that also going way in v8?

PR #83 (2021-07-01): Remove & operator when passing display driver to lv_disp_flush_ready
Closes #81
commit fa86e1d0 (2021-07-01): Remove & operator when passing display driver to lv_disp_flush_ready

Closes #81