lvgl/lvgl_esp32_drivers #85: On ESP32C3 with ILI9488 controller at 480x320 16bit, the DMA memory is not sufficient for an entire screen to flush

dastarling, 2021-07-07:
The function ili9488_flush() is using an algorithm that allocates an entire screen sized DMA buffer, but on the ESP32C3 device, there is not sufficient memory.

I modified the function to do one line at a time, but it might be good to figure out the best of allocation.

I have a modification to this driver that I can share, but it seems that many of the drivers share this flushing code.

Perhaps there is a way to double check the memory required and do the entire screen if available, otherwise drop back to one line or x lines at a time?

dastarling, 2021-07-07:
Perhaps break out common code from the drivers to make the code more maintainable?  Might be a larger task, but likely worth it.

C47D, 2021-07-08:
> Perhaps there is a way to double check the memory required and do the entire screen if available, otherwise drop back to one line or x lines at a time?

This might be a good solution.

> Perhaps break out common code from the drivers to make the code more maintainable? Might be a larger task, but likely worth it.

I think it worth the effort to make all ILIx drivers similar, I don't really know their differences, my first guess is the resolution they can handle. Any thoughts @kisvegabor @embeddedt @tore-espressif ?

wreyford, 2021-07-09:
I hope it will get attention, as I have run into similar memory issues and that on ESP32WROOM32E and then crashes.
I converted from LVGL7.1 to 8.1 hoping that it would improve the issue, but to no avail. I have a medium sized project, and have run out of memory for flushing buffer.

dastarling, 2021-07-10:
I seem to be so short on memory that I may need to modify the driver to 8 bits as well?

tore-espressif, 2021-07-12:
> Perhaps there is a way to double check the memory required and do the entire screen if available, otherwise drop back to one line or x lines at a time?

I'd prefer not 'guessing' how much memory the driver can allocate; my only argument is:
- Not deterministic. The buffer size would depend on current free heap size, which can lead to unpredictable behavior

I'd like to keep it simple stupid: let's the application programmer decide on his memory allocation - he should know the best.
He'd just pass two integers (size of buffer 0 and 1) and that would be enough.

Any thoughts?

C47D, 2021-07-29:
@tore-espressif Are you suggesting using the `user_data` field on the display driver?

@dastarling Can you share your patch?

tore-espressif, 2021-08-02:
So the only reason we allocate another intermediate buffer https://github.com/lvgl/lvgl_esp32_drivers/blob/master/lvgl_tft/ili9488.c#L130 is to extrapolate LVGL's 16bit color-depth to the display's 18bit color-depth set in the initialization https://github.com/lvgl/lvgl_esp32_drivers/blob/master/lvgl_tft/ili9488.c#L65.

I'd suggest setting the display color-depth to 16 bit so we can optimize the flush function by removing the intermediate buffer.
ILI9481 and 9488 drivers could be merged.
Unfortunately I don't have ILI9481/9488 on my desk, is there anyone willing to try the fixed version?

(I should have checked the datasheet before commenting, sry).

dastarling, 2021-08-09:
Sorry for not getting back sooner.  I have found a combination of memory buffers that works most of the time, but reviewing the data sheet for the ILI9488 I think  I would concur that modifying the driver to use the 16 bit color format directly would eliminate the extra buffer and return the memory for other usages.
I had a difficult time squeezing in the graphics and MBED TLS with all the other requirements.  Now the memory buffers are tiny and I suspect performance will be slower.
I'll take a look at the ILI9481 and see if we could just use it if it supports the 16bit color directly instead of converting to 18bit.
I can also test any proposed changes on the ILI9488.  It appears the IM configuration is set to '111' on my display.

dastarling, 2021-08-09:
Actually from studying the datasheets, it looks like the ILI9341 supports the 565 bit format, the ILI9481 driver appears to be the same as the ILI9488. The datasheet shows it supports the 565 format, but does not provide an example of that combination.

dastarling, 2021-08-09:
After trying to output 565 via SPI, it does not seem to work, which is probably why the datasheet is inconsistent.  See discussion here: https://www.avrfreaks.net/forum/ili9488-spi-mode-xmega, so I guess we are stuck with 18bit  transforms

tore-espressif, 2021-08-10:
> so I guess we are stuck with 18bit transforms

oh my, that's bad news.

But we sure need to get rid of that intermediate buffer: I'd always transform 1 pixel and send it over SPI (3 bytes). The performance will be poor, but memory requirements a lot better.

ILI9481/9488 would then be a great candidate for our planned parallel interface.

tore-espressif, 2021-10-04:
Closing this as 'won't fix'. See details here: https://github.com/lvgl/lvgl_esp32_drivers/pull/103#issuecomment-933508835

closed 2021-10-04 by tore-espressif