lvgl/lvgl_esp32_drivers #90: SSD1306 orientation

michox, 2021-07-20:
I would be thrilled if it would be possible to implement 90 and 270 degree rotations (i.e. Portrait mode) for the SSD1306 driver. 

I would love to use LVGL for my project but I would need this capability. I tried to see if I could implement it myself but failed. It should be fairly simple to implement using some conversions and a switch statement at the last place before sending the command to the display. 

If somebody could find some time in the near future to help me implement that, I would be very grateful!

tore-espressif, 2021-08-03:
SSD1306 does not support portrait orientation, so the rotation must be in software. Will have a look probably this week

tore-espressif, 2021-08-25:
Actually, LVGL v8 has changed handling of display rotation. So we can start working on it after we have support for v8.

adam2809, 2022-02-01:
Hi I just found this issue while trying to figure out portrait mode for SSD1306 on esp32 and I just wanted to ask if there are any updates. Please let me know about the progress and if the issue is still unresolved I'd love to help since I need this for a project.

adam2809, 2022-02-01:
Thanks for the quick response @C47D do you have any examples of the round_px implementations which have this functionality? I am using lvgl 7.11.0 and lvgl_esp32_drivers form commit f3f6ba64. I tried to set up newer varsions of these repos but this is the only version combination that I was able to get working although I read in some threads that in lvgl v8 will help with getting portrait so I am considering an upgrade.

C47D, 2022-02-02:
@adam2809 Sorry I don't have an example at hand, I think I saw it being mentioned over the forum. I will take a look over there tomorrow after work.

adam2809, 2022-02-02:
@C47D I have no idea why I didn't just use the versions from `lv_port_esp32` but I just tried it and it works perfectly. The solution I tried so far was setting my `lv_disp_drv_t` like this:
`disp_drv.sw_rotate = 1;`
`disp_drv.rotated = LV_DISP_ROT_90; // or LV_DISP_ROT_270`
but this just results in random artifacts that display only on one side of the screen.

C47D, 2022-02-02:
Hi, the rotations in the repo were introduced before the "official" rotation support and they don't line up. In the particular case of you driver you have two rotation options in the menuconfig configuration page, landscape and inverted landscape. Both portrait orientations doesn't work.

Working on lining up both rotation approaches has been in my to do list for quite a while, I just haven't had the time to work on it.