lvgl/lvgl_esp32_drivers #91: Cant Use Touch with ILI9488 3,5" 480x320 V1.0 XPT2046

Trackhe, 2021-07-29:
Describe the issue
I cant use my Touch.

Code to reproduce the issue
I use the arduino esp port. And load the Library TFT_eSPI
the kallibration of the touch is working after that i init the lvgl and run a loop
it also works with my ILI9341 2.8 240x320 V1.2 with the same Touch controller XPT2046

void loop1code(void * pvParameters) {
for(;;){
lv_task_handler();
vTaskDelay(10 / portTICK_PERIOD_MS);
}
}

framework: espressif

installed components
arduino
lvgl
lvgl_esp32_drivers

when i configure the lvgl_esp32_drivers (touch controller) in menuconfig take this no effekt.
same problem.

i have aktivated the cpu count in the bottom right corner but it is not showing. up. maybe this is not a touch problem? I dont know how i can test it? With the ili9341 it works.

Expected Results
working touch

Actual Results
no touch. but compiling and flashing suscessfull

ESP32 Chip version
ESP: M213EH2864UH3Q0 https://www.mouser.de/ProductDetail/Espressif-Systems/ESP32-WROVER-IEM213EH2864UH3Q0?qs=vmHwEFxEFR97Z2otHlKj0Q%3D%3D

ESP-IDF version
4.4

Development machine OS
macos

Compilation warnings/errors (if available)
no

If possible, copy the compilation log into a file and attach it here
if needed

tore-espressif, 2021-08-03:
there is no TFT_eSPI library in this repo, so I guess you mean this one https://www.arduino.cc/reference/en/libraries/tft_espi/

If you want to use LVGL + touch you have to use display/touch drivers from this repo only.

Trackhe, 2021-08-03:
@tore-espressif you dont understand my issue. I wanted to say that the display works with TFT_eSPI from Boomer (no hardware failure). And with the touch from the 3.5" dp dosnt work then i use the display/touch driver. That is the issue.

tore-espressif, 2021-08-04:
Could you please post a link to your code, output log and your schematic?

I can't really help you with the information that you provided

Trackhe, 2021-08-04:
main code
```
#include <stdbool.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_freertos_hooks.h"
#include "freertos/semphr.h"
#include "esp_system.h"
#include "esp_log.h"
#include "driver/gpio.h"
#include "esp_sleep.h"
#include "soc/rtc.h"
#include "soc/sens_reg.h"
#include "soc/rtc_periph.h"
#include "driver/rtc_io.h"

#include "lvgl.h"
#include "lvgl_helpers.h"

static void lv_tick_task(void *arg);

TaskHandle_t Task1;
static void loop1code(void *pvParameters);

extern "C" void app_main(void)
{
    ESP_LOGI(TAG, "Initializing");

  xTaskCreatePinnedToCore(
      loop1code, /* Function to implement the task */
      "na1", /* Name of the task */
      4096*2,  /* Stack size in words */
      NULL,  /* Task input parameter */
      0,  /* Priority of the task */
      &Task1,  /* Task handle. */
      1); /* Core where the task should run */
}

SemaphoreHandle_t xGuiSemaphore;

static void loop1code(void *pvParameters) {

    xGuiSemaphore = xSemaphoreCreateMutex();

    lv_init();
    lvgl_driver_init();

    lv_color_t* buf1 = (lv_color_t*)heap_caps_malloc(DISP_BUF_SIZE * sizeof(lv_color_t), MALLOC_CAP_DMA);
    assert(buf1 != NULL);
    static lv_color_t *buf2 = NULL;
    
    static lv_disp_buf_t disp_buf;

    uint32_t size_in_px = DISP_BUF_SIZE;

    lv_disp_buf_init(&disp_buf, buf1, buf2, size_in_px);

    lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);
    disp_drv.flush_cb = disp_driver_flush;

    disp_drv.buffer = &disp_buf;
    lv_disp_drv_register(&disp_drv);

    /* Register Input #Touch */
    #if CONFIG_LV_TOUCH_CONTROLLER != TOUCH_CONTROLLER_NONE
[... 831 lines cut ...]
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options
```

i use actually the version 7.11 lvgl.
and if i use the 2.8" dp with 320x240 same pins. but another dp controller ILI9341 it works all normaly.
It is worth mentioning that when I display the cpu / fps display, the ili9488 only shows a still image or not the display at all. with the ili9341 everything works. Both have the same touch controller.

SinglWolf, 2021-09-24:
> and when i touch i get an interrupt but not the correct position. it seems to be always 0 maybe a defekt touch controller?

```
I (37199) XPT2046: P(0,0)
I (37199) XPT2046: P_norm(0,0)
I (37199) XPT2046: x = 480, y = 320
```
I have the same problem. The likelihood of a defekt touch controller has halved.

Trackhe, 2021-09-25:
> The likelihood of a defekt touch controller has halved.

I test it with another library to ensure that the controller works.

SinglWolf, 2021-09-25:
> I test it with another library to ensure that the controller works.

If it works, can you share a working example?

Trackhe, 2021-09-25:
> > I test it with another library to ensure that the controller works.
> 
> 
> 
> If it works, can you share a working example?

It works but not with lvgl, that is the reason why the issue exist. But when you don't want to use lvgl or you want to test your touch chip use Bodmer/TFT_eSPI on github.

SinglWolf, 2021-09-25:
> It works but not with lvgl, that is the reason why the issue exist. But when you don't want to use lvgl or you want to test your touch chip use Bodmer/TFT_eSPI on github.

That's why I ask for a working example, uploaded the sketch, and you're done. I'm not strong in Arduino.

SinglWolf, 2021-09-26:
A demo sketch from [Bodmer/TFT_eSPI](https://github.com/Bodmer/TFT_eSPI) for touchscreen calibration works great (chip ESP32-S2)!
We are waiting for a solution to the problem from the developers.

SinglWolf, 2021-09-26:
Meanwhile... The solution was found by itself!
While setting up a demo sketch from `Bodmer/TFT_eSPI` to calibrate the touchscreen, I came across a warning like this:
```c
#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)
```
Dear developers. Nail down and paint the above warning red!
The question can be closed. This is victory!

Trackhe, 2021-09-26:
That is the solution. Nice. I wonder how i can miss that. Thank you so mouch. Is the SDO from Display needed? Because it works also when i left them disconnected. Nice work!

closed 2021-09-26 by Trackhe