lvgl/lvgl_esp32_drivers #98: support of parallel displays 16/8-bit

Rajssss, 2021-08-08:
Is there any plan for adding parallel display support?
moreover and can we add support?
I am thinking a separate component to handle 8/16-bit I/Os, like disp_i2s_parallel. We can add support there, and modify the display drivers accordingly, also adding config support for SPI and 8/16 bit selection.

tore-espressif, 2021-08-09:
Yes, we would like to add more graphical interfaces.

But developing and testing them here is rather difficult, because they can be considered _peripheral drivers_ rather then LVGL specific drivers. Initial plan is to abstract the graphical interface away and use peripheral drivers maintained by Espressif https://github.com/espressif/esp-idf/tree/master/components/esp_lcd

If you want to experiment with LCD parallel interface on Espressif chip, `esp_lcd` might be a good place to start.

To this date, the parallel driver works only on ESP32-S3, update for ESP32 and ESP32-S2 should be merged soon.

closed 2021-08-16 by Rajssss

modi12jin, 2021-08-17:
@Rajssss https://github.com/espressif/esp-iot-solution

I asked a question which has some details

```
https://github.com/espressif/esp-iot-solution/issues/122
```