lvgl/lvgl_esp32_drivers #104: ST7789 Static Noise displaying to screen during initialization

tvanfossen, 2021-08-31:
ST7789 initialization seems out of date.

In LVGL @ master, LVGL v7 and LVGL v8 the same issue is present, where static noise is displayed to the screen during initialization in st7789.c

I've tried a few different changes to the command list with no luck.

The driver commands seem out of date with documentation found here.
https://www.rhydolabz.com/documents/33/ST7789.pdf

There are a slew of commands being past to the init that do not exist in the documentation as far as I can tell, and ones that have no associated define in st7789.h. With these undefined commands commented out, the problem of static noise on the screen remains.

Any help would be much appreciated! The static noise kinda kills this driver for the project I am working on, as it shows on every bootup, and in this case the display is on/off often

C47D, 2021-09-01:
I think I have that display at hand, let me confirm that and I will let you know.

On the meantime, can you upload the `sdkconfig.h` file available in `build/config`? You can attach it to your comment.
Also, how are you wiring the display with the MCU and what MCU are you using?

tore-espressif, 2021-09-01:
@tvanfossen can you control LCD's backlight from the MCU?

The easiest way is to enable the backlight only after you have transferred color data to the display, so there is always valid data in the Graphics RAM

tvanfossen, 2021-09-01:
Co-worker was able to get the issue resolved inside of st7789.c - will either let him update or get the fix from him

MCU does not have backlight control, that was the first idea for a quick fix

On a custom board with an esp32, I dont have the wiring diagram at hand

[sdkconfig.zip](https://github.com/lvgl/lvgl_esp32_drivers/files/7093189/sdkconfig.zip)

C47D, 2021-09-05:
Hi @tvanfossen , I've found my display and ESP32 kit, and I'm able to flash the demo, I'm not able to see the static noise (I'm not using the custom display buffer size tho), are you using a buyed display?

My pc is pretty slow, it takes about 15mins to build a project, so I will be patient to try find the issue 😅

C47D, 2021-09-05:
So I've used the custom display buffer size and I'm not seeing any issues, here's my sdkconfig file in case you want to take a look at it. And a bit from the monitor output:

```
I (406) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (30) lvgl_helpers: Display hor size: 240, ver size: 240
I (40) lvgl_helpers: Display buffer size: 4096
I (40) lvgl_helpers: Initializing SPI master for display
I (50) lvgl_helpers: Configuring SPI host VSPI_HOST (2)
I (50) lvgl_helpers: MISO pin: -1, MOSI pin: 23, SCLK pin: 18, IO2/WP pin: -1, IO3/HD pin: -1
I (60) lvgl_helpers: Max transfer size: 8192 (bytes)
I (70) lvgl_helpers: Initializing SPI bus...
I (70) disp_spi: Adding SPI device
I (70) disp_spi: Clock speed: 20000000Hz, mode: 2, CS pin: -1
ST7789 initialization.
I (480) st7789: Display orientation: PORTRAIT
I (480) st7789: 0x36 command value: 0xC0
```

[sdkconfig.txt](https://github.com/lvgl/lvgl_esp32_drivers/files/7112080/sdkconfig.txt)

jparker324, 2021-09-07:
@C47D I work with @tvanfossen.
In order to clear the display memory, I've had to replace [Lines 78-80](https://github.com/lvgl/lvgl_esp32_drivers/blob/dd09b4d01afcccd3c876fc464ef86e0da4cd03be/lvgl_tft/st7789.c#L78) of the st7789.c file :
```c
{ST7789_CASET, {0x00, 0x00, 0x00, 0xEF}, 4},
{ST7789_RASET, {0x00, 0x00, 0x01, 0x3f}, 4},
{ST7789_RAMWR, {0}, 0},
```
with a longer sequence of memory writes
```c
{ST7789_RASET, {0x00, 0x00, 0x00, 0xEF}, 4},
{ST7789_CASET, {0x00, 0x00, 0x00, 0x10}, 4},
{ST7789_RAMWR, {0}, 0},
{ST7789_RASET, {0x00, 0x00, 0x00, 0xEF}, 4},
{ST7789_CASET, {0x00, 0x11, 0x00, 0x21}, 4},
{ST7789_RAMWR, {0}, 0},
{ST7789_RASET, {0x00, 0x00, 0x00, 0xEF}, 4},
{ST7789_CASET, {0x00, 0x22, 0x00, 0x32}, 4},
{ST7789_RAMWR, {0}, 0},
{ST7789_RASET, {0x00, 0x00, 0x00, 0xEF}, 4},
{ST7789_CASET, {0x00, 0x33, 0x00, 0x43}, 4},
{ST7789_RAMWR, {0}, 0},
{ST7789_RASET, {0x00, 0x00, 0x00, 0xEF}, 4},
{ST7789_CASET, {0x00, 0x44, 0x00, 0x54}, 4},
{ST7789_RAMWR, {0}, 0},
{ST7789_RASET, {0x00, 0x00, 0x00, 0xEF}, 4},
{ST7789_CASET, {0x00, 0x55, 0x00, 0x65}, 4},
{ST7789_RAMWR, {0}, 0},
{ST7789_RASET, {0x00, 0x00, 0x00, 0xEF}, 4},
{ST7789_CASET, {0x00, 0x66, 0x00, 0x76}, 4},
{ST7789_RAMWR, {0}, 0},
{ST7789_RASET, {0x00, 0x00, 0x00, 0xEF}, 4},
{ST7789_CASET, {0x00, 0x77, 0x00, 0x87}, 4},
{ST7789_RAMWR, {0}, 0},
{ST7789_RASET, {0x00, 0x00, 0x00, 0xEF}, 4},
{ST7789_CASET, {0x00, 0x88, 0x00, 0x98}, 4},
{ST7789_RAMWR, {0}, 0},
{ST7789_RASET, {0x00, 0x00, 0x00, 0xEF}, 4},
{ST7789_CASET, {0x00, 0x99, 0x00, 0xA9}, 4},
{ST7789_RAMWR, {0}, 0},
{ST7789_RASET, {0x00, 0x00, 0x00, 0xEF}, 4},
{ST7789_CASET, {0x00, 0xAA, 0x00, 0xBA}, 4},
{ST7789_RAMWR, {0}, 0},
{ST7789_RASET, {0x00, 0x00, 0x00, 0xEF}, 4},
{ST7789_CASET, {0x00, 0xBB, 0x00, 0xCB}, 4},
{ST7789_RAMWR, {0}, 0},
{ST7789_RASET, {0x00, 0x00, 0x00, 0xEF}, 4},
{ST7789_CASET, {0x00, 0xCC, 0x00, 0xDC}, 4},
{ST7789_RAMWR, {0}, 0},
{ST7789_RASET, {0x00, 0x00, 0x00, 0xEF}, 4},
{ST7789_CASET, {0x00, 0xDD, 0x00, 0xED}, 4},
{ST7789_RAMWR, {0}, 0},
{ST7789_RASET, {0x00, 0x00, 0x00, 0xEF}, 4},
{ST7789_CASET, {0x00, 0xEE, 0x00, 0xFE}, 4},
{ST7789_RAMWR, {0}, 0},
{ST7789_RASET, {0x00, 0x00, 0x00, 0xEF}, 4},
{ST7789_CASET, {0x00, 0xFF, 0x01, 0x0F}, 4},
{ST7789_RAMWR, {0}, 0},
{ST7789_RASET, {0x00, 0x00, 0x00, 0xEF}, 4},
{ST7789_CASET, {0x01, 0x10, 0x01, 0x20}, 4},
{ST7789_RAMWR, {0}, 0},
{ST7789_RASET, {0x00, 0x00, 0x00, 0xEF}, 4},
{ST7789_CASET, {0x01, 0x21, 0x01, 0x31}, 4},
{ST7789_RAMWR, {0}, 0},
{ST7789_RASET, {0x00, 0x00, 0x00, 0xEF}, 4},
{ST7789_CASET, {0x01, 0x32, 0x01, 0x3f}, 4},
{ST7789_RAMWR, {0}, 0},
```
and then add some handling for the `ST7789_RAMWR` command, such that the data payload is taken from a 4080 bytes buffer filled with the desired default color.

For this, I replaced the code around [Line 109-118](https://github.com/lvgl/lvgl_esp32_drivers/blob/dd09b4d01afcccd3c876fc464ef86e0da4cd03be/lvgl_tft/st7789.c#L109)  with the following.
```c
#define DISP_BUF_SIZE 4080

lv_color_t color_map[DISP_BUF_SIZE];
for(uint16_t i=0; i < DISP_BUF_SIZE; i++)
{
    color_map[i].full = 0xA310;  //grey color
    // color_map[i].full = 0x0000;  //black (memset would be faster)
}

//Send all the commands
uint16_t cmd = 0;
while (st7789_init_cmds[cmd].databytes!=0xFF) {
    if (st7789_init_cmds[cmd].cmd != ST7789_RAMWR)
    {
        st7789_send_cmd(st7789_init_cmds[cmd].cmd);
        st7789_send_data(st7789_init_cmds[cmd].data, st7789_init_cmds[cmd].databytes&0x1F);
        if (st7789_init_cmds[cmd].databytes & 0x80) {
                vTaskDelay(100 / portTICK_RATE_MS);
        }
    }
    else
    {
        st7789_send_cmd(st7789_init_cmds[cmd].cmd);
        st7789_send_data((void *)&color_map, DISP_BUF_SIZE * 2);
    }
    cmd++;
}
```
I suppose it may be further simplified with `ST7789_RAMWRC` commands but I haven't tried that yet.

And not to detract from the core issue, but can you explain the purpose of the commands in `st7789_init_cmds` that aren't enumerated?  See [Line 54](https://github.com/lvgl/lvgl_esp32_drivers/blob/dd09b4d01afcccd3c876fc464ef86e0da4cd03be/lvgl_tft/st7789.c#L54) for example.  The commands aren't defined in any ST7789 driver documentation I can find and removing them seems to have no effect.

jparker324, 2021-09-07:
@C47D  Alternatively, I can also initialize the display successfully by sending the full address range once (as the driver currently does), but then use repeated `ST7789_RAMWRC` commands with the previously mentioned 4080 byte color data payload
```c
{ST7789_RASET, {0x00, 0x00, 0x00, 0xEF}, 4},
{ST7789_CASET, {0x00, 0x00, 0x01, 0x3F}, 4},
{ST7789_RAMWR, {0}, 0},
{ST7789_RAMWRC, {0}, 0},
{ST7789_RAMWRC, {0}, 0},
{ST7789_RAMWRC, {0}, 0},
{ST7789_RAMWRC, {0}, 0},
{ST7789_RAMWRC, {0}, 0},
{ST7789_RAMWRC, {0}, 0},
{ST7789_RAMWRC, {0}, 0},
{ST7789_RAMWRC, {0}, 0},
{ST7789_RAMWRC, {0}, 0},
{ST7789_RAMWRC, {0}, 0},
{ST7789_RAMWRC, {0}, 0},
{ST7789_RAMWRC, {0}, 0},
{ST7789_RAMWRC, {0}, 0},
{ST7789_RAMWRC, {0}, 0},
{ST7789_RAMWRC, {0}, 0},
{ST7789_RAMWRC, {0}, 0},
{ST7789_RAMWRC, {0}, 0},
{ST7789_RAMWRC, {0}, 0},
```

C47D, 2021-09-07:
Hi @jparker324 , thanks for reaching out, I understand your issue now, you mean the "noise" when the display is powered up, with the code you sent you're "clearing" the display before initializing it, so once it starts up it dosn't show the noise, isn't?

Which of your approaches feels more correct to you? I think we could initialize the `color_map` array to 0 with this:

```c
#define DISP_BUF_SIZE 4080

lv_color_t color_map[DISP_BUF_SIZE] = {0};
```

> And not to detract from the core issue, but can you explain the purpose of the commands in st7789_init_cmds that aren't enumerated? See Line 54 for example. The commands aren't defined in any ST7789 driver documentation I can find and removing them seems to have no effect.

To be honest I don't remember where those values came from, we can remove them if they aren't necessary.

C47D, 2021-09-08:
It does look good to me, are you planning to submit your changes into a PR? I would only rename the `DISP_BUF_SIZE` symbol, afaik it's not related to the size allocated for the LVGL work buffer, isn't? I'm also working on #92 , which is using the same display driver, if you need me to test your changes please let me know.