lvgl/lvgl_esp32_drivers #111: Functions for controlling the brightness of the display backlight. Ideas and suggestions.

SinglWolf, 2021-09-18:
Sorry for the many symbols.
I was pleased that a built-in function for controlling the brightness of the display backlight appeared.
But when trying to integrate the disp_backlight_set () function into my project, I ran into an unsolvable problem for me.
For the disp_backlight_set () function to work properly, a pointer to an initialized disp_backlight_h is required. Initialization occurs inside the function void * disp_driver_init (void), inside the function disp_backlight_h bckl_handle is also declared.
Inside the function void * disp_driver_init (void) there is a return bckl_handle, but that doesn't give us anything, because (void) in Africa is also (void).
In the beginning, the idea was to suggest that you change the function type void * disp_driver_init (void) to disp_backlight_h * disp_driver_init (void) to get a pointer when initializing the display.
On reflection, I came to the conclusion: all this is worthless.
There are only three channels for PWM. We cannot change the channel number for the display, it is always numbered #0. And for me this channel is used to reproduce sound signals. I do not foresee two display, as for the majority of users of your wonderful library.
To begin with, I propose to bring the backlight brightness control functions to a working state and simplify them to control only one display.

My edits:
`components\lvgl_esp32_drivers\lvgl_tft\Kconfig`
```diff
@@ -956,6 +956,20 @@ menu "LVGL TFT Display controller"
             help
                 Display backlight is controlled by pulse-width modulation, allowing brightness settings.
 
+        config LV_DISP_BACKLIGHT_PWM_CHANNEL
+            int "PWM channel" if LV_DISP_BACKLIGHT_PWM
+            default 0
+            range 0 3
+            help
+                The channel number for controlling the brightness of the display backlight.
+
+        config LV_DISP_BACKLIGHT_PWM_BRIGHTNESS_PERCENT
+            int "Brightness display backlight" if LV_DISP_BACKLIGHT_PWM
+            default 100
+            range 0 100
+            help
+                Display brightness as a percentage.
+
     endchoice
 
     config LV_BACKLIGHT_ACTIVE_LVL
```
Result:
[image]
`lvgl_esp32_drivers/lvgl_tft/disp_driver.c`
```diff
@@ -7,7 +7,7 @@
 #include "esp_lcd_backlight.h"
 #include "sdkconfig.h"
 
-void *disp_driver_init(void)
+void disp_driver_init(void)
 {
 #if defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_ILI9341
     ili9341_init();
@@ -61,13 +61,10 @@ void *disp_driver_init(void)
         .output_invert = true, // Backlight on low
 #endif
         .timer_idx = 0,
-        .channel_idx = 0 // @todo this prevents us from having two PWM controlled displays
+        .channel_idx = CONFIG_LV_DISP_BACKLIGHT_PWM_CHANNEL
     };
-    disp_backlight_h bckl_handle = disp_backlight_new(&bckl_config);
-    disp_backlight_set(bckl_handle, 100);
-    return bckl_handle;
-#else
-    return NULL;
+    disp_backlight_new(&bckl_config);
+    disp_backlight_set(CONFIG_LV_DISP_BACKLIGHT_PWM_BRIGHTNESS_PERCENT);
 #endif
 }
```
`lvgl_esp32_drivers/lvgl_tft/disp_driver.h`
```diff
@@ -67,7 +67,7 @@ extern "C" {
  **********************/
 
 /* Initialize display */
-void *disp_driver_init(void);
+void disp_driver_init(void);
 
 /* Display flush callback */
 void disp_driver_flush(lv_disp_drv_t * drv, const lv_area_t * area, lv_color_t * color_map);
```
`lvgl_esp32_drivers/lvgl_tft/esp_lcd_backlight.c`
```diff
@@ -17,21 +17,23 @@ typedef struct {
     int index;        // Either GPIO or LEDC channel
 } disp_backlight_t;
 
+disp_backlight_t *bckl_dev = NULL;
+
 static const char *TAG = "disp_backlight";
 
-disp_backlight_h disp_backlight_new(const disp_backlight_config_t *config)
+void disp_backlight_new(const disp_backlight_config_t *config)
 {
     // Check input parameters
     if (config == NULL)
-        return NULL;
+        return;
     if (!GPIO_IS_VALID_OUTPUT_GPIO(config->gpio_num)) {
         ESP_LOGW(TAG, "Invalid GPIO number");
-        return NULL;
+        return;
     }
-    disp_backlight_t *bckl_dev = calloc(1, sizeof(disp_backlight_t));
+    bckl_dev = calloc(1, sizeof(disp_backlight_t));
     if (bckl_dev == NULL){
         ESP_LOGW(TAG, "Not enough memory");
-        return NULL;
+        return;
     }
 
     if (config->pwm_control){
@@ -67,20 +69,18 @@ disp_backlight_h disp_backlight_new(const disp_backlight_config_t *config)
         gpio_matrix_out(config->gpio_num, SIG_GPIO_OUT_IDX, config->output_invert, false);
     }
 
-    return (disp_backlight_h)bckl_dev;
 }
 
-void disp_backlight_set(disp_backlight_h bckl, int brightness_percent)
+void disp_backlight_set(int brightness_percent)
 {
     // Check input paramters
-    if (bckl == NULL)
+    if (bckl_dev == NULL)
         return;
     if (brightness_percent > 100)
         brightness_percent = 100;
     if (brightness_percent < 0)
         brightness_percent = 0;
 
-    disp_backlight_t *bckl_dev = (disp_backlight_t *) bckl;
     ESP_LOGI(TAG, "Setting LCD backlight: %d%%", brightness_percent);
 
     if (bckl_dev->pwm_control) {
@@ -92,16 +92,15 @@ void disp_backlight_set(disp_backlight_h bckl, int brightness_percent)
     }
 }
 
-void disp_backlight_delete(disp_backlight_h bckl)
+void disp_backlight_delete(void)
 {
-    if (bckl == NULL)
+    if (bckl_dev == NULL)
         return;
 
-    disp_backlight_t *bckl_dev = (disp_backlight_t *) bckl;
     if (bckl_dev->pwm_control) {
         ledc_stop(LEDC_LOW_SPEED_MODE, bckl_dev->index, 0);
     } else {
         gpio_reset_pin(bckl_dev->index);
     }
-    free (bckl);
+    free (bckl_dev);
 }
```
`lvgl_esp32_drivers/lvgl_tft/esp_lcd_backlight.h`
```diff
@@ -18,12 +18,6 @@ extern "C" { /* extern "C" */
  * GLOBAL PROTOTYPES
  **********************/
 
-/**
- * @brief Display backlight controller handle
- *
- */
-typedef void * disp_backlight_h;
-
 /**
  * @brief Configuration structure of backlight controller
  *
@@ -46,7 +40,7 @@ typedef struct {
  * @param[in] config Configuration structure of backlight controller
  * @return           Display backlight controller handle
  */
-disp_backlight_h disp_backlight_new(const disp_backlight_config_t *config);
+void disp_backlight_new(const disp_backlight_config_t *config);
 
 /**
  * @brief Set backlight
@@ -57,8 +51,8 @@ disp_backlight_h disp_backlight_new(const disp_backlight_config_t *config);
  * @param bckl                   Backlight controller handle
  * @param[in] brightness_percent Brightness in [%]
  */
-void disp_backlight_set(disp_backlight_h bckl, int brightness_percent);
-void disp_backlight_delete(disp_backlight_h bckl);
+void disp_backlight_set(int brightness_percent);
+void disp_backlight_delete(void);
 
 #ifdef __cplusplus
 } /* extern "C" */
```
If you like my idea, I will be glad.
There will be an incentive to develop the idea of using multiple display drivers in one project.
Once again I apologize for such a long text.

C47D, 2021-09-18:
Hi, I'm in a road travel right now, but we're working on improving all the drivers on the `develop` branch.

I've separated the backlight initialization from the driver initialization function, but that branch will be changing a lot for a couple of weeks (a.k.a not ready for projects).

Please let me know if the work we're doing there is something along the lines of your patches.

Regards
- carlos

SinglWolf, 2021-09-18:
> Please let me know if the work we're doing there is something along the lines of your patches.

```c
#endif

    return disp_backlight_init();
}

void *disp_backlight_init(void)
{
```

This separation of the backlight initialization function from the driver initialization function is really just a formality.
What's the point of separating the backlight initialization function this way? When exiting the driver initialization function, the backlight initialization function anyway called. In my opinion, this approach only complicates the understanding of the code.
In the rest of the code, alas, I did not see anything similar to my edits.

tore-espressif, 2021-09-19:
Hello @SinglWolf 

Thank you very much for this issue. When we merged the backlight function, we left few TODOs, so it's good that you pointed them out. Most importantly, there is no good example how to use the new API (my bad).

We definitely want to go with multi-display support, so I'll fix the issues you mentioned in that manner.

SinglWolf, 2021-09-19:
> We definitely want to go with multi-display support, so I'll fix the issues you mentioned in that manner.

Multiple display support? On different controllers? Will my dreams and aspirations come true?

closed 2021-09-20 by SinglWolf