lvgl/lvgl_esp32_drivers #113: Support for multiple displays on different controllers and more. Ideas and Reflections.

SinglWolf, 2021-09-20:
The essence of the problem. It's global to me. On the verge of a complete rejection of "LVGL" completely.
I will give an example (purely theoretically). The SPI bus frequency is 40 MHz and cannot be changed by the end user. The connected display can only operate at 35 MHz. The owner of the display should ask the firmware developer to compile a custom firmware for their display.
Another example (already practical). There is a device with 100% software compatibility. But it is incompatible with the display pins assigned in the firmware. Is it necessary for the owner of the device to cut the tracks on the board and reconnect the display as indicated in the firmware? Or contact the firmware developer again? Or order a printed circuit board for this firmware?
Most likely, the device user will look for another solution. There are many of them on the Internet.
And the developer of this ill-fated firmware?
I don’t know about others, but I’m trying to change something for the better.
The idea is this. Give the developer users of your driver library the ability to change the settings of certain parameters at runtime, that is, on the fly.
How? Need to invent something? Change the concept completely?
Yes and no.
Look at how the settings system is arranged in the "LVGL" itself.
There it is possible to disable the Kconfig settings and use your own configuration file. Why don't You go the same way?
If you like my idea, I can provide more specific examples of how to create such a configuration file in terms of pin reassignment and some other points.
Thank you for reading such a long opus.
Sorry, it's boiling.

tore-espressif, 2021-09-20:
>  But it is incompatible with the display pins assigned in the firmware. 

You can set all pins in menuconfig, if you don't use any predefined display. `Select predefined display configuration` set to `None`.

> Is it necessary for the owner of the device to cut the tracks on the board and reconnect the display as indicated in the firmware? Or contact the firmware developer again? Or order a printed circuit board for this firmware?

I believe you are not serious with these questions.

> I don’t know about others, but I’m trying to change something for the better.

You can help by submitting meaningful Pull Requests. Issues like this doesn't help at all.

> Now even in components\lvgl_esp32_drivers\CMakeLists.txt it is written...

We have recently pointed you to `develop` branch. Check it out here: 
https://github.com/lvgl/lvgl_esp32_drivers/blob/develop/CMakeLists.txt

> likely they are not compiled

Everything that is added to CMake configuration is compiled. Linker takes care of elimination of unused symbols.

> extra modules in some incomprehensible way fall into the firmware file

Linker behavior is deterministic.

> An int is allocated under brightness_percent

Nothing is allocated here. This value is passed as a parameter to the function via CPU register. CPU registers are 32bit wide.

> uint32_t duty_cycle - sane? No. Need uint8_t duty_cycle, because the result of the calculation will never exceed 100.

Answer to your question is in a comment, directly after the statement. If you read it, you wouldn't be so confused.
`uint32_t duty_cycle = (1023 * brightness_percent) / 100; // LEDC resolution set to 10bits, thus: 100% = 1023`
`uint16_t` would suffice here, but `ledc_set_duty` accepts `uint32_t` so we are using this type.

> But in my project, I was able to allocate only 1024 bytes for the display buffer

you can set display buffer size in menuconfig.

You have made lot of false assumption about C language, compiling, linking and running software on embedded target.
As you can see, it is not trivial. The multidisplay support, in particular, is difficult to add with current state of the code. We made few steps to get there, but there is also on option to start from the scratch.

closed 2021-09-20 by tore-espressif

SinglWolf, 2021-09-20:
> You have made lot of false assumption about C language, compiling, linking and running software on embedded target.
You didn't even pay attention to my idea of using an independent config. Of course, I'm a dumb C user who just mastered "Hello World!" trying to teach a C programming guru.
Yes, God be with You. Write whatever You want. Good luck.

embeddedt, 2021-09-20:
I would like to remind everyone to please be constructive when providing suggestions/ideas for improvement. Statements like "you didn't even pay attention" and "do you care about reducing compile times?" are not actionable and do not really follow our code of conduct either.

I am not that familiar with the ESP32 project in its current state, as it's not a platform I use, but I think that @tore-espressif and @C47D are genuinely trying to redesign the architecture to be as useful as possible, and the bikeshedding in this post is not particularly constructive. There may be some good suggestions here, but they are difficult to pick out.

Please do not take this statement personally. Feel free to open more specific issue(s) on aspects you feel need improvement. But, at the moment, I am locking this topic.