lvgl/lvgl_esp32_drivers #114: Independent use of i2c_manager.

SinglWolf, 2021-09-20:
Why is `i2c_manager` only available when using a supported display?
I already have an I2C-controlled devices in my project. It is an audio processor and a real time clock. A separate component is used for this. I would like to use the `lvgl_esp32_drivers/lvgl_i2c` component, but it is not available due to unsupported the display.
Can't `i2c_manager` be enabled independently of the display?
Or am I again having a false of the `C` language?

C47D, 2021-09-20:
Maybe @ropg can give us a hint?

ropg, 2021-09-20:
Just use the I2C Manager as a separate component and use it to talk to your I2C devices, and then tell LLVGL to use its locking so they all play nice. I think it's pretty well explained in the [README](https://github.com/ropg/i2c_manager), but I'd be happy to tweak the text.

Help me understand: 

* You have a project. It has some other parts that talk I2C: RTC and audio.
* Your display is 'unsupported'. By LVGL? Or what do you mean there?
* You'd like to use I2C Manager to talk to RTC and audio.

Right?

Maybe some description of hardware, software and intentions would help.

SinglWolf, 2021-09-21:
In my naive simplicity and misconception about the C language, I assumed that the I2C manager settings in Kconfig were not available to me due to the fact that my display was not SSD1306.
After reading your post, full of confusion, I selected the SSD1306 display via menuconfig to see how things really are.
Shock and Awe. There is not even a way to assign pins for I2C.
Your I2C Manager as a component even for the SSD1306 display (which is officially recognized as supported by lvgl_esp32_drivers) is not properly integrated into the lvgl_esp32_drivers library.
I close my issue, since the continuation of the discussion does not make sense. The developer is always right.

closed 2021-09-21 by SinglWolf

ropg, 2021-09-21:
But, but...

Within lvgl_esp32_drivers the I2C settings for a given port are potentially shared by a screen and a touch device. That's why you can only set which I2C port a sensor or screen is on and there are no pin settings there. The actual I2C port settings, including pins, are the next item below "LVGL ESP Drivers", called "I2C Port Settings". Unless something is wrong, in which case I'd like to investigate. Because I do not believe developers are always right, and bugs and weird corner cases happen.

kisvegabor, 2021-09-21:
@SinglWolf 
> The developer is always right.

That's not true. @ropg asked 3 questions to better understand your issue in order to help. I can't see answers to them but you just some insulting wording.  

The community of LVGL should be composed of positive, supportive, and respectful people.  Please, keep it in mind to have constructive and productive discussion in which people are happy to participate.

SinglWolf, 2021-09-21:
> Within lvgl_esp32_drivers the I2C settings for a given port are potentially shared by a screen and a touch device. That's why you can only set which I2C port a sensor or screen is on and there are [...]

Settings for any device should be simple and intuitive. I hope it's hard to disagree with this.
There are two displays with different controllers. If the controllers are different, this is not a reason to do different setting methods for them. Otherwise, everything goes head over heels.
I initially had the ILI9488 display selected, there is no I2C Port Settings line in the menu.
[image]
In the display settings of ILI9488 there is a menu for setting the pins of the SPI bus.
[image]
[image]
When the SSD1306 display is selected, the display pin settings menu is not available. Port number only.
[image]
Should I have known that the ill-fated I2C pin settings would appear in a completely different place, several levels higher?
[image]
Either I have to know, or I'm as stupid as a cork and wasting precious time from developers for some nonsense.

> I can't see answers to them just some insulting words.

Do you find my words offensive? Sorry, I wasn't going to offend anyone. Not now, not in the past. In the future, I also do not plan any insults to anyone's address of the team.
Out of my foolishness I dared to ask questions, propose an idea, point out some mistakes (and what a shame, I myself was mistaken) and received nonsense in a mentoring tone and insults in return. This is fine?

ropg, 2021-09-21:
The way I2C works in the drivers is documented, [here](https://github.com/lvgl/lvgl_esp32_drivers/tree/master/lvgl_i2c#readme), but you absolutely have a point that it could be more obvious. So no, you're not stupid. (Just a little quick in assuming we're unhelpful maybe.) Not sure how to make it obvious other than to stick it in the help for the Kconfig options, but honestly most people don't press '?' to look there. I'm very open to additional ways to let users know.

( Before I incorporated I2C Manager there was I2C code all over the drivers, and a convoluted bunchof ifdefs to try and make sure nothing conflicted if two devices needed I2C. The problem of sharing resources between drivers (and even different ESP-IDF components) that it solves is unfortunately non-trivial. The way it works for SPI is a bit like the old code for I2C worked and it's problematic for the same reason: you can set pins in multiple places, but there's only one port in hardware. Without one place to set things like pins it's always going to be a hack. )

SinglWolf, 2021-09-21:
> Not sure how to make it obvious other than to stick it in the help for the Kconfig options, but honestly most people don't press '?' to look there. I'm very open to additional ways to let users know.

And what will happen if you make the setting of the I2C pins and the SPI bus in a separate (general) menu item? Then i2c_manager as a component for other devices can be enabled.