lvgl/lvgl_esp32_drivers #115: kconfig for lvgl_esp32_drivers/lvgl_i2 causes ESP-IDF SDK Configuration Editor to Crash

wreyford, 2021-09-21:
I've updated to the latest master of lvgl_esp32_drivers. 
I'm using LVGL8.3 master.
[image]
I can select i2C port Zero, but in the KConfig specific to I2C Port Settings, 
if I select Enable I2C port 0 or Enable I2C port 1, instead of bringing up the options to set the pincs etc, it just crashes menuconfig.

Please, I don't know how to set my pins.

Also, when including this line of code as per the readme before calling lv_init():
 ``` 
lvgl_i2c_locking(i2c_manager_locking()); 
```

I get the following compiler error:
> implicit declaration of function 'i2c_manager_locking'; did you mean 'i2c_master_write'? [-Werror=implicit-function-declaration]

wreyford, 2021-09-22:
In the sdkconfig, the following [ CONFIG_LV_I2C=y and CONFIG_LV_I2C_DISPLAY_PORT=0] appears to be set, but it is not selected anywhere in menuconfig:

'''
# CONFIG_LV_DISP_BACKLIGHT_OFF is not set
CONFIG_LV_DISP_BACKLIGHT_SWITCH=y
# CONFIG_LV_DISP_BACKLIGHT_PWM is not set
# CONFIG_LV_BACKLIGHT_ACTIVE_LVL is not set
CONFIG_LV_DISP_PIN_BCKL=-1
CONFIG_LV_I2C=y
CONFIG_LV_I2C_DISPLAY_PORT=0
# end of LVGL TFT Display controller
'''

and in the kconfig file for lvgl_esp32_drivers/lvgl_tft/kconfig:

line 996
'''
config LV_I2C
        bool
        default y if LV_I2C_DISPLAY

    config LV_I2C_DISPLAY_PORT
        int
        default 1 if LV_I2C_DISPLAY_PORT_1
        default 0
'''

And  at line
177

'''
 # Display controller communication protocol
    #
    # This symbols define the communication protocol used by the
    # ESP32 to communicate with the display controller.
    # This symbols can be used, but not limited, to:
    # - Know what peripherals to initialize.
    # - Know if the touch and display controllers share the same peripheral.
    # - Etc.
    config LV_TFT_DISPLAY_PROTOCOL_SPI
        bool
        help
            Display controller protocol SPI

    config LV_I2C_DISPLAY
        bool
        help
            Display controller protocol I2C
'''
But I'm using the ILI9488, and thus 
'''
config LV_TFT_DISPLAY_USER_CONTROLLER_ILI9488
        bool "ILI9488"
            select LV_TFT_DISPLAY_CONTROLLER_ILI9488
            select LV_TFT_DISPLAY_PROTOCOL_SPI
'''
but this does not exclusively set SPI, so LV_I2C=y is also set????

wreyford, 2021-09-25:
I resorted to screen recording the error that flashes on the screen momentarily in VSCode using the ESP-IDF plugin, when the menuconfig crashes.
The error was a result of the int value not having a default, and thus '' 
Fixing this allowed for the proper menuconfig handling of the I2C ports

```
if I2C_MANAGER_0_ENABLED
		config I2C_MANAGER_0_SDA
			int "SDA (GPIO pin)"
			default 0
		config I2C_MANAGER_0_SCL
			int "SCL (GPIO pin)"
			default 0
```
and this change is required for both ports 0 and 1.

Now all I need to do is get past the 
**implicit declaration of function 'i2c_manager_locking'; did you mean 'i2c_master_write'? [-Werror=implicit-function-declaration]** error

tore-espressif, 2021-09-27:
@ropg could you please have a look?

ropg, 2021-09-27:
Will do as soon as I find some time.

C47D, 2021-10-16:
Ok, let's close this, feel free to open it again if you have issues.

closed 2021-10-16 by C47D