lvgl/lvgl_esp32_drivers #116: Strange artifacts on the ILI9488 display

SinglWolf, 2021-09-23:
I already broken my head over this problem.
Strange vertical stripes appeared on the display.
Explain, please, did I broken something in the settings, or is it a sign of the imminent demise of the display?
[image]
`ESP32-S2` device, `ILI9488` display, VScode + PlatformIO environment, Windows 10 system. Environment components are the freshest.
```c
/***************************
 * CURRENT VERSION OF LVGL
 ***************************/
#define LVGL_VERSION_MAJOR 7
#define LVGL_VERSION_MINOR 9
#define LVGL_VERSION_PATCH 1
#define LVGL_VERSION_INFO "dev"
```
I just don't remember when (under which version of `lvgl_esp32_drivers` or framework `esp-idf`) there were no artifacts, there was a big break in working with the project.
I can't roll back, I would have known that there would be such a problem - I would have made a commit.

SinglWolf, 2021-09-23:
After consistently downgrading the `espidf` framework versions, I found a version (`3.2.1` according to `PlatformIO` numbering), where there are no artifacts. But in this version there is a bug - `UART` does not work, `ESP32-S2` just hangs. In the next one (`3.3.0` according to `PlatformIO` numbering) and the current one, `UART` works, but there are artifacts on the display.
Should I move to `lvgl/lv_platformio` with this question or stay here?

tore-espressif, 2021-09-24:
Hello again :)
I'll check this on HW as soon as I get it.

Meanwhile to your other questions:
If you are using `esp-idf` then ESP32-S2 is supported from v4.2 onwards.
ATM, I'm not sure whether this is PlatformIO related... once I get it working we can debug this further.

I  can see from your activity on GitHub that you really want to make LVGL happen in your projects, I'm sorry that it is not in a state that would enable you to use full potential for LVGL (and ESPs).

We are currently working on making this better.
If you're interested, please contact me directly at tomas.rezucha@espressif.com and describe requirements of your project, ie. chip/board you are using + LCDs that you intend to use, or any other relevant requirement. If we could prepare an example that would exactly fit your needs, we could kill two birds with one stone.

Looking forward to hearing from you,
Tomas Rezucha

SinglWolf, 2021-09-27:
After I managed to solve the problem with the touchscreen, I decided to try the fresh version `LVGL`. I started with an example from the archive controlPanel.zip that I took [here](https://github.com/lvgl/lvgl_esp32_drivers/issues/68#issuecomment-864407634).
I have successfully migrated the project to PlatformIO and launched it. There are no artifacts.
However, I spent a lot of time synchronizing the coordinates of the touchscreen with the orientation of the display. But this is a separate topic for discussion.
It's time to move on, I thought, and I updated both LVGL and the drivers in the project.
First try.
Setting display backlight without PWM - I get a black screen..
Setting display backlight with PWM - I get a white screen.
And that's all.
Second try.
I returned the drivers from the archive. The example display appeared, but again with artifacts.
[image]

I am in complete despair, confusion and upset feelings.

SinglWolf, 2021-11-07:
I was able to completely get rid of the artifacts on the display screen with the `ILI9488` controller using very old drivers.
Here is from my project set of functions  which there are no artifacts on the display.:
```c
void disp_spi_add_device(spi_host_device_t host)
{
    gpio_num_t cs;
    gpio_get_lcd(&cs, NULL, NULL);
    ESP_LOGI(TAG, "Adding SPI device");
    ESP_LOGI(TAG, "Clock speed: %dHz, mode: %d, CS pin: %d",
             SPI_TFT_CLOCK_SPEED_HZ, SPI_TFT_SPI_MODE, cs);

    spi_device_interface_config_t devcfg_disp = {
        .clock_speed_hz = SPI_TFT_CLOCK_SPEED_HZ,
        .mode = 0,
        .spics_io_num = cs, // CS pin
        .queue_size = 1,
        .post_cb = spi_ready,
        .flags = SPI_DEVICE_NO_DUMMY | SPI_DEVICE_HALFDUPLEX,
    };

    // Attach the LCD to the SPI bus
    esp_err_t ret = spi_bus_add_device(host, &devcfg_disp, &spi_disp);
    assert(ret == ESP_OK);
}

void tp_spi_add_device(spi_host_device_t host)
{
    gpio_num_t t_cs;
    esp_err_t ret = gpio_get_touch(&t_cs, NULL);
    if (t_cs == GPIO_NONE)
        return;

    gpio_config_t cs_config = {
        .pin_bit_mask = BIT64(t_cs),
        .mode = GPIO_MODE_OUTPUT,
        .pull_up_en = GPIO_PULLUP_DISABLE,
        .pull_down_en = GPIO_PULLDOWN_DISABLE,
        .intr_type = GPIO_INTR_DISABLE,
    };

    ret = gpio_config(&cs_config);
    assert(ret == ESP_OK);
    spi_device_interface_config_t devcfg_touch = {
        .command_bits = 8,
        .clock_speed_hz = 2000000,
        .mode = 0,
        .spics_io_num = t_cs,
        .queue_size = 1,
        .pre_cb = NULL,
        .post_cb = NULL,
    };

    // Attach the Touch to the SPI bus
    ret = spi_bus_add_device(host, &devcfg_touch, &spi_touch);
    assert(ret == ESP_OK);
}

void disp_spi_send_data(uint8_t *data, uint16_t length)
{
    if (length == 0)
        return;

    while (spi_trans_in_progress)
        ;

    memset(&transaction, 0, sizeof(transaction));
    transaction.length = length * 8; // transaction length is in bits
    transaction.tx_buffer = data;

    spi_trans_in_progress = true;
    spi_color_sent = false;
    ESP_ERROR_CHECK(spi_device_queue_trans(spi_disp, &transaction, portMAX_DELAY));
}

void disp_spi_send_colors(uint8_t *data, uint16_t length)
{
    if (length == 0)
        return;

    while (spi_trans_in_progress)
        ;

    memset(&transaction, 0, sizeof(transaction));
    transaction.length = length * 8; // transaction length is in bits
    transaction.tx_buffer = data;

    spi_trans_in_progress = true;
    spi_color_sent = true;
    ESP_ERROR_CHECK(spi_device_queue_trans(spi_disp, &transaction, portMAX_DELAY));
}

uint16_t tp_spi_xchg_xpt(uint8_t writeData)
{
    while (spi_trans_in_progress)
        ;

    uint8_t data_recv[2] = {0, 0};
    uint16_t data_send = 0;

    memset(&transaction, 0, sizeof(transaction));
    transaction.cmd = writeData;
    transaction.length = 16; // length is in bits
    transaction.tx_buffer = &data_send;
    transaction.rx_buffer = &data_recv;

    ESP_ERROR_CHECK(spi_device_queue_trans(spi_touch, &transaction, portMAX_DELAY));

    spi_transaction_t *rt;
    ESP_ERROR_CHECK(spi_device_get_trans_result(spi_touch, &rt, portMAX_DELAY));

    return (uint16_t)(data_recv[0]) << 8 | (uint16_t)(data_recv[1]);
}

void disp_tp_spi_finished()
{
    while (spi_trans_in_progress)
        ;
}

/**********************
 *   STATIC FUNCTIONS
 **********************/

static void IRAM_ATTR spi_ready(spi_transaction_t *trans)
{

    if (spi_color_sent)
    {
        spi_color_sent = false;
        spi_trans_in_progress = false;
        lv_disp_t *disp = NULL;

        disp = _lv_refr_get_disp_refreshing();
        lv_disp_flush_ready(disp->driver);
    }
    else
    {
        spi_trans_in_progress = false;
    }
}
```

SinglWolf, 2022-01-19:
It is unfortunate that the problem of artifacts on the display screen is of no interest to anyone.

dlugaz, 2022-03-01:
I've encoutered the same issue (with artifacts). I think the artifacts are there, because the last pixel of the buffer didnt get one of the bytes of color sent.
I think the size of mybuf should be 3*size + 1, at least in my case it made the artifacts dissapear. 
```

 uint32_t size = lv_area_get_width(area) * lv_area_get_height(area);

    lv_color16_t *buffer_16bit = (lv_color16_t *) color_map;
    uint8_t *mybuf;
    do {
        mybuf = (uint8_t *) heap_caps_malloc(3 * size * sizeof(uint8_t), MALLOC_CAP_DMA);
        if (mybuf == NULL) {
            LV_LOG_WARN("Could not allocate enough DMA memory!");
        }
    } while (mybuf == NULL);

    uint32_t LD = 0;
    uint32_t j = 0;

    for (uint32_t i = 0; i < size; i++) {
        LD = buffer_16bit[i].full;
        mybuf[j] = (uint8_t) (((LD & 0xF800) >> 8) | ((LD & 0x8000) >> 13));
        j++;
        mybuf[j] = (uint8_t) ((LD & 0x07E0) >> 3);
        j++;
        mybuf[j] = (uint8_t) (((LD & 0x001F) << 3) | ((LD & 0x0010) >> 2));
        j++;
    }

```