lvgl/lvgl_esp32_drivers #118: ft6x36.c error in code results in touch not giving correct co-ordinates solved.

wreyford, 2021-09-27:
I ran into a problem when I updated to master branch using LVGL8.3
Amongst many issues, this was my last stumbling block.

The touch was giving incorrect coordinates, making it unusable.
For my ILI9488, using capacitive touch over I2C, with kconfig:
```
#
# Touchpanel Configuration (FT6X06)
#
CONFIG_LV_FT6X36_SWAPXY=y
# CONFIG_LV_FT6X36_INVERT_X is not set
CONFIG_LV_FT6X36_INVERT_Y=y
# end of Touchpanel Configuration (FT6X06)
```

I got incorrect X values

The reason for this is, that in the code line 118 to 136:
```    last_x = ((data_buf[1] & FT6X36_MSB_MASK) << 8) | (data_buf[2] & FT6X36_LSB_MASK);
    last_y = ((data_buf[3] & FT6X36_MSB_MASK) << 8) | (data_buf[4] & FT6X36_LSB_MASK);

#if CONFIG_LV_FT6X36_INVERT_X
    last_x =  LV_HOR_RES - last_x;
#endif
#if CONFIG_LV_FT6X36_INVERT_Y
    last_y = LV_VER_RES - last_y;
#endif
#if CONFIG_LV_FT6X36_SWAPXY
    int16_t swap_buf = last_x;
    last_x = last_y;
    last_y = swap_buf;
#endif
    data->point.x = last_x;
    data->point.y = last_y;
    data->state = LV_INDEV_STATE_PR;
    ESP_LOGD(TAG, "X=%u Y=%u", data->point.x, data->point.y);
    return false;```

needs to be:
```   last_x = ((data_buf[1] & FT6X36_MSB_MASK) << 8) | (data_buf[2] & FT6X36_LSB_MASK);
    last_y = ((data_buf[3] & FT6X36_MSB_MASK) << 8) | (data_buf[4] & FT6X36_LSB_MASK);

#if CONFIG_LV_FT6X36_SWAPXY
    int16_t swap_buf = last_x;
    last_x = last_y;
    last_y = swap_buf;
#endif
#if CONFIG_LV_FT6X36_INVERT_X
    last_x =  LV_HOR_RES - last_x;
#endif
#if CONFIG_LV_FT6X36_INVERT_Y
    last_y = LV_VER_RES - last_y;
#endif
    data->point.x = last_x;
    data->point.y = last_y;
    data->state = LV_INDEV_STATE_PR;
    ESP_LOGI(TAG, "X=%u Y=%u", data->point.x, data->point.y);
    return false;```

As you can see you must first swap the XY, before inverting, or if inverting y then swopping it, results in you incorrectly inverting x. Now the LV_HOR_RES or LV_VERT_RES is incorrect for that axis, and get crazy values!

My proposed fix fixes the issue for me.
Please check and fix if you agree.

C47D, 2021-09-28:
Thanks for the detailed information, I think I don't have that particular controller to test your fix. But I will try to add it in the coming days.

Regards.

wreyford, 2021-09-28:
Thx, if you look at the actual logic, You don't want to invert an axis before swopping it, as the end user expects to invert a specific axis, x or y, and doing that after swopping XY, results in the expected axis being swopped and then obviously deducted from the correct screen coordinate for width and height of display.

C47D, 2021-09-29:
Does 8862804 fixes the issue?

wreyford, 2021-09-29:
Yes thanks, it does.

C47D, 2021-09-29:
Cool, I'll close the issue then.

closed 2021-09-29 by C47D