lvgl/lvgl_esp32_drivers #122: LV_HOR_RES_MAX not compatible with LVGL v8

morganrallen, 2021-10-01:
_MAX has been dropped.

A snippet like this should suffice to support v8 and maintain reverse compatibility.
```
diff --git a/lvgl_helpers.h b/lvgl_helpers.h
index 5fd6f09..90b92ed 100644
--- a/lvgl_helpers.h
+++ b/lvgl_helpers.h
@@ -34,6 +34,10 @@ extern "C" {
  * color format being used, for RGB565 each pixel needs 2 bytes.
  * When using the mono theme, the display pixels can be represented in one bit,
  * so the buffer size can be divided by 8, e.g. see SSD1306 display size. */
+#ifndef LV_HOR_RES_MAX
+#define LV_HOR_RES_MAX LV_HOR_RES
+#endif
+
 #if defined (CONFIG_CUSTOM_DISPLAY_BUFFER_SIZE)
 #define DISP_BUF_SIZE   CONFIG_CUSTOM_DISPLAY_BUFFER_BYTES
 #else
```

This appears to be the last issue to getting v8 compatibility, for at least the FT6X36 driver.

Thanks!

kisvegabor, 2021-10-01:
In v8 `lv_disp_get_hor_res(NULL)` should be used instead.

morganrallen, 2021-10-04:
I see. What do you think best course of action is then to clean up all the `#define` statements using `_MAX` in `lvgl_helpers.h`, `lvgl_spi_conf.h`, `st7735` and `EVE_config.h`?

kisvegabor, 2021-10-06:
You replace `LV_HOR_RES_MAX` with the "hor_res" definition of the TFT.

We are working on rewriting the drivers and they work with v8.

wreyford, 2021-12-16:
Any progress on this incompatibility matter?
We value all your hard work and commitment.

C47D, 2021-12-23:
Hi,

Thanks for the words, I've been particularly busy with my day job, we're trying to get into a position where we can start the support for v8, there are several things that changed since v7.

C47D, 2022-01-08:
Should be fixed with #161

closed 2022-01-08 by C47D