lvgl/lvgl_esp32_drivers #124: Migration from version 7 to version 8 - Guru Meditation Error...

SinglWolf, 2021-10-03:
I started migrating my project from `LVGL` version 7 to version 8. I commented out all the lines with compiler errors in order to make the firmware and then gradually put the project in order.
Display controller `ILI9488`.
And I ran into a problem where it shouldn't be in principle. My project in version 7 works fine. In the place where the problem arose, I did not make any changes, because in version 8 nothing has changed. Here is a code snippet from the project where the crash occurred:
```c
// ..........................
	LOGO = lv_img_create(lv_scr_act());
#if defined(CONFIG_IDF_TARGET_ESP32S2)
	LV_IMG_DECLARE(RadiolaS2);
	lv_img_set_src(LOGO, &RadiolaS2);
#elif defined(CONFIG_IDF_TARGET_ESP32)
	LV_IMG_DECLARE(radiola2);
	lv_img_set_src(LOGO, &radiola2);
#endif
	lv_obj_align(LOGO, LV_ALIGN_CENTER, 0, 0);
#endif
	lv_refr_now(NULL); // <----- Here ESP32-C2 crash
// ..........................
```
The picture loads normally and crashes:
```
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x401a6462  PS      : 0x00060031  A0      : 0x8002842e  A1      : 0x3ffccc60
A2      : 0x3ffdf110  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3ffdf100
A6      : 0x00060723  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x3ffcc4d0  A11     : 0x3f4c2084  A12     : 0x00000006  A13     : 0x00000014
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000018  LBEG    : 0x00000006  LEND    : 0x00000014  LCOUNT  : 0x40027ce9
Core  0 was running in ISR context:
EPC1    : 0x401078c7  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace:0x401a645f:0x3ffccc60 0x4002842b:0x3ffccc80 0x40026a3f:0x3ffccca0 0x40026b99:0x3ffcccc0 0x40027e06:0x3ffcccf0 0x401ac897:0x3ffdf820 0x401a32ef:0x3ffdf840 0x40030afd:0x3ffdf860 0x40032161:0x3ffdf880
  #0  0x401a645f:0x3ffccc60 in lv_disp_flush_ready at components\lvgl\src\hal/lv_hal_disp.c:366
  #1  0x4002842b:0x3ffccc80 in spi_ready at components\lvgl_esp32_drivers\lvgl_tft/disp_spi.c:350
  #2  0x40026a3f:0x3ffccca0 in spi_post_trans at C:\Users\Alexander\.platformio\packages\framework-espidf\components\driver/spi_master.c:584 (discriminator 1)
  #3  0x40026b99:0x3ffcccc0 in spi_intr at C:\Users\Alexander\.platformio\packages\framework-espidf\components\driver/spi_master.c:618
  #4  0x40027e06:0x3ffcccf0 in _xt_lowint1 at C:\Users\Alexander\.platformio\packages\framework-espidf\components\freertos\port\xtensa/xtensa_vectors.S:1105
  #5  0x401ac897:0x3ffdf820 in cpu_ll_waiti at C:\Users\Alexander\.platformio\packages\framework-espidf\components\hal\esp32s2\include/hal/cpu_ll.h:202
      (inlined by) esp_pm_impl_waiti at C:\Users\Alexander\.platformio\packages\framework-espidf\components\esp_pm/pm_impl.c:827
  #6  0x401a32ef:0x3ffdf840 in esp_vApplicationIdleHook at C:\Users\Alexander\.platformio\packages\framework-espidf\components\esp_common\src/freertos_hooks.c:63
  #7  0x40030afd:0x3ffdf860 in prvIdleTask at C:\Users\Alexander\.platformio\packages\framework-espidf\components\freertos/tasks.c:3839 (discriminator 1)
  #8  0x40032161:0x3ffdf880 in vPortTaskWrapper at C:\Users\Alexander\.platformio\packages\framework-espidf\components\freertos\port\xtensa/port.c:168

ELF file SHA256: 2db741b074703b07

CPU halted.
```
Why do I think it is crashing on `lv_refr_now(NULL)`?
I commented out all the lines with `lv_refr_now(NULL)` in the project and ESP32-S2 continued to work further exactly until the moment when the task to update the display content was triggered:
```c
#if defined(CONFIG_IDF_TARGET_ESP32S2)
		if ((MainConfig->lcd_type != LCD_NONE))
		{
			if (xSemaphoreTake(xGuiSemaphore, portMAX_DELAY))
			{

				lv_task_handler(); // <------  main/services.c:1005
				xSemaphoreGive(xGuiSemaphore);
			}
		}
#endif
```
With an error like this:
```
I (15857) SERVICES: task_lcd Started, LCD Type: 1
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x80030bf7  PS      : 0x00060830  A0      : 0x80098535  A1      : 0x3fff7520  
A2      : 0x3fff7540  A3      : 0x80030bf7  A4      : 0x3fff7530  A5      : 0x0000000f
A6      : 0x00000000  A7      : 0x40032028  A8      : 0x8009846c  A9      : 0x3fff7500  
A10     : 0x3ffdf144  A11     : 0x00000026  A12     : 0x3ffd2918  A13     : 0x0000000f
A14     : 0x00fb0793  A15     : 0x00fb0b70  SAR     : 0x0000001d  EXCCAUSE: 0x00000014  
EXCVADDR: 0x80030bf4  LBEG    : 0x3ffd2918  LEND    : 0x0000000f  LCOUNT  : 0x40027ce9

Backtrace:0x40030bf4:0x3fff7520 0x40098532:0x3fff7540 0x4009c199:0x3fff7580 0x4009d683:0x3fff75b0 0x4009bff3:0x3fff75d0 0x4009ca71:0x3fff75f0 0x4009cd38:0x3fff7640 0x4009cd1c:0x3fff7660 0x4009cd1c:0x3fff7680 0x4009cd1c:0x3fff76a0 0x4009cda0:0x3fff76c0 0x400a0649:0x3fff76e0 0x400aa199:0x3fff7710 0x400aa242:0x3fff7730 0x4008a59d:0x3fff7750 0x40032161:0x3fff7780
  #0  0x40030bf4:0x3fff7520 in vTaskDelay at C:\Users\Alexander\.platformio\packages\framework-espidf\components\freertos/tasks.c:1526
  #1  0x40098532:0x3fff7540 in lv_event_send at components\lvgl\src\core/lv_event.c:73
  #2  0x4009c199:0x3fff7580 in lv_obj_get_self_width at components\lvgl\src\core/lv_obj_pos.c:587
  #3  0x4009d683:0x3fff75b0 in lv_obj_get_scroll_right at components\lvgl\src\core/lv_obj_scroll.c:235
  #4  0x4009bff3:0x3fff75d0 in calc_auto_size at components\lvgl\src\core/lv_obj_pos.c:931
  #5  0x4009ca71:0x3fff75f0 in lv_obj_refr_size at components\lvgl\src\core/lv_obj_pos.c:107
  #6  0x4009cd38:0x3fff7640 in layout_update_core at components\lvgl\src\core/lv_obj_pos.c:956
  #7  0x4009cd1c:0x3fff7660 in layout_update_core at components\lvgl\src\core/lv_obj_pos.c:949 (discriminator 3)
  #8  0x4009cd1c:0x3fff7680 in layout_update_core at components\lvgl\src\core/lv_obj_pos.c:949 (discriminator 3)
  #9  0x4009cd1c:0x3fff76a0 in layout_update_core at components\lvgl\src\core/lv_obj_pos.c:949 (discriminator 3)
  #10 0x4009cda0:0x3fff76c0 in lv_obj_update_layout at components\lvgl\src\core/lv_obj_pos.c:306
  #11 0x400a0649:0x3fff76e0 in _lv_disp_refr_timer at components\lvgl\src\core/lv_refr.c:200
  #12 0x400aa199:0x3fff7710 in lv_timer_exec at components\lvgl\src\misc/lv_timer.c:315 (discriminator 2)
  #13 0x400aa242:0x3fff7730 in lv_timer_handler at components\lvgl\src\misc/lv_timer.c:106
  #14 0x4008a59d:0x3fff7750 in lv_task_handler at components\lvgl/src/lv_api_map.h:37
      (inlined by) task_lcd at main/services.c:1005
  #15 0x40032161:0x3fff7780 in vPortTaskWrapper at C:\Users\Alexander\.platformio\packages\framework-espidf\components\freertos\port\xtensa/port.c:168

ELF file SHA256: e40ede10bedc232b

CPU halted.
```
Please help me figure out the problem.
Recently, all my messages have not been answered.
What is it? Conspiracy of silence?
So just let me know so.
Whereupon I'm leaving to look for another solution for displaying graphics for my project. And I will not waste my time on empty correspondence here.

C47D, 2021-10-03:
I haven't tested using LVGLv8, I will get some ESP32S2 and C3 kits next week to try test the drivers.

From the first backtrace it seems to me that the `disp->driver` might be NULL, here https://github.com/lvgl/lvgl_esp32_drivers/blob/f6999a52fe6ab7a7e4d035e0d96c0cfb1021a94a/lvgl_tft/disp_spi.c#L316, can you validate it isn't NULL?

SinglWolf, 2021-10-04:
I Inserted check lines into the function.
Didn't work even once.
```c
static void IRAM_ATTR spi_ready(spi_transaction_t *trans)
{
    disp_spi_send_flag_t flags = (disp_spi_send_flag_t)trans->user;

    if (flags & DISP_SPI_SIGNAL_FLUSH)
    {
        lv_disp_t *disp = NULL;

#if (LVGL_VERSION_MAJOR >= 7)
        disp = _lv_refr_get_disp_refreshing();
#else /* Before v7 */
        disp = lv_refr_get_disp_refreshing();
#endif

#if LVGL_VERSION_MAJOR < 8
        lv_disp_flush_ready(&disp->driver);
#else
        if (disp->driver == NULL) // <---- check
            printf("disp->driver == NULL\n"); // <---- check
        lv_disp_flush_ready(disp->driver); // <---- disp_spi.c:352
#endif
    }

    if (chained_post_cb)
    {
        chained_post_cb(trans);
    }
}
```
log:
```
I (4857) DISPLAY: Display buffer size: 1024
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x401a6456  PS      : 0x00060031  A0      : 0x80028445  A1      : 0x3ffccc70
A2      : 0x3ffdf100  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3ffdf110
A6      : 0x00060723  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000001
A10     : 0x3ffcc4dc  A11     : 0x3f4c2084  A12     : 0x00000006  A13     : 0x00000014
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000019  LBEG    : 0x00000006  LEND    : 0x00000014  LCOUNT  : 0x40027cf1
Core  0 was running in ISR context:
EPC1    : 0x401078f7  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace:0x401a6453:0x3ffccc70 0x40028442:0x3ffccc90 0x40026a47:0x3ffcccb0 0x40026ba1:0x3ffcccd0 0x40027e0e:0x3ffccd00 0x401ac88b:0x3ffdf830 0x401a32e3:0x3ffdf850 0x40030b15:0x3ffdf870 0x40032179:0x3ffdf890
  #0  0x401a6453:0x3ffccc70 in lv_disp_flush_ready at components\lvgl\src\hal/lv_hal_disp.c:366
  #1  0x40028442:0x3ffccc90 in spi_ready at components\lvgl_esp32_drivers\lvgl_tft/disp_spi.c:352
  #2  0x40026a47:0x3ffcccb0 in spi_post_trans at C:\Users\Alexander\.platformio\packages\framework-espidf\components\driver/spi_master.c:584 (discriminator 1)
  #3  0x40026ba1:0x3ffcccd0 in spi_intr at C:\Users\Alexander\.platformio\packages\framework-espidf\components\driver/spi_master.c:618
  #4  0x40027e0e:0x3ffccd00 in _xt_lowint1 at C:\Users\Alexander\.platformio\packages\framework-espidf\components\freertos\port\xtensa/xtensa_vectors.S:1105
  #5  0x401ac88b:0x3ffdf830 in cpu_ll_waiti at C:\Users\Alexander\.platformio\packages\framework-espidf\components\hal\esp32s2\include/hal/cpu_ll.h:202
      (inlined by) esp_pm_impl_waiti at C:\Users\Alexander\.platformio\packages\framework-espidf\components\esp_pm/pm_impl.c:827
  #6  0x401a32e3:0x3ffdf850 in esp_vApplicationIdleHook at C:\Users\Alexander\.platformio\packages\framework-espidf\components\esp_common\src/freertos_hooks.c:63
  #7  0x40030b15:0x3ffdf870 in prvIdleTask at C:\Users\Alexander\.platformio\packages\framework-espidf\components\freertos/tasks.c:3839 (discriminator 1)
  #8  0x40032179:0x3ffdf890 in vPortTaskWrapper at C:\Users\Alexander\.platformio\packages\framework-espidf\components\freertos\port\xtensa/port.c:168

ELF file SHA256: a64599c5e4374413

CPU halted.
```
If I understood Your question correctly, of course.

SinglWolf, 2021-10-05:
I knew that the lv_disp_draw_buf_t structure in version 8 should be static, but I missed it.
I sprinkle ashes on my head.

closed 2021-10-05 by SinglWolf