lvgl/lvgl_esp32_drivers #159: ESP32C3 SPI configuration

C47D, 2021-12-27:
Well, this story is still not finished yet :(
As of [ESP32-C3 datasheet v1.1](https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf#page=20) (page 20): 
```
ESP32-C3 features three SPI interfaces (SPI0, SPI1, and SPI2). 
SPI0 and SPI1 can only be configured to operate in SPI memory mode,
while SPI2 can be configured to operate in both SPI memory and general-purpose SPI modes.
```
As of [online documentation on ESP32-C3](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/spi_master.html):
```
ESP32-C3 integrates 3 SPI peripherals.
SPI0 and SPI1 are used internally to access the ESP32-C3’s attached flash memory. 
Both controllers share the same SPI bus signals, and there is an arbiter to determine 
which can access the bus. Currently, SPI Master driver does not support SPI1 bus.
```
So at least there is no such thing as SPI3 and `SPI3_HOST` for ESP32-C3, though online docs have this somehow:
```
enum spi_host_device_t
Enum with the three SPI peripherals that are software-accessible in it.
Values:
SPI1_HOST = 0
SPI1.
SPI2_HOST = 1
SPI2.
SPI3_HOST = 2
SPI3.
```
As of [spi_types.h from esp-idf](https://github.com/espressif/esp-idf/blob/master/components/hal/include/hal/spi_types.h) there is no such thing as SPI0 or SPI0_HOST.

And definitely SPI1 can't work as a general purpose SPI controller now.

Sorry for not testing all the cases, but please don't revert this PR.

_Originally posted by @arktrin in https://github.com/lvgl/lvgl_esp32_drivers/issues/145#issuecomment-1001219264_

C47D, 2021-12-27:
The invalid host message when calling `spi_bus_initialize` comes from `is_valid_host`, for C3 `SOC_SPI_PERIPH_NUM` value is 2, so I would say using `SPI2_HOST` would be safe, isn't @arktrin 

```c
static inline bool is_valid_host(spi_host_device_t host)
{
#if (SOC_SPI_PERIPH_NUM == 2)
    return host >= SPI1_HOST && host <= SPI2_HOST;
#elif (SOC_SPI_PERIPH_NUM == 3)
    return host >= SPI1_HOST && host <= SPI3_HOST;
#endif
}
```

arktrin, 2021-12-27:
SPI2_HOST is working ok without any problem.

arktrin, 2021-12-27:
@C47D I've already posted [this issue in esp-idf](https://github.com/espressif/esp-idf/issues/8144). Will it be helpful or no chance to be noticed in reasonable time? or people in charge are already here?

arktrin, 2022-02-03:
@C47D  Since it is a esp-idf issue and not directly corresponds to `lvgl_esp32_drivers` - it may be closed.

C47D, 2022-02-03:
Thanks @arktrin for the update.

closed 2022-02-03 by C47D