lvgl/lvgl_esp32_drivers #168: [develop] Compilation errors in esp_lcd_backlight.c

C47D, 2022-01-11:
@C47D Thanks a lot! But with a latest develop now I've got even larger number of errors:
```
../main/main.c: In function 'guiTask':
../main/main.c:137:16: warning: unused variable 'display' [-Wunused-variable]
     lv_disp_t* display = lv_disp_drv_register(&disp_drv);
                ^~~~~~~

../components/lvgl_esp32_drivers/lvgl_tft/esp_lcd_backlight.c: In function 'disp_backlight_new':
../components/lvgl_esp32_drivers/lvgl_tft/esp_lcd_backlight.c:59:9: error: implicit declaration of function 'gpio_matrix_out'; did you mean 'gpio_iomux_out'? [-Werror=implicit-function-declaration]
         gpio_matrix_out(config->gpio_num, ledc_periph_signal[LEDC_LOW_SPEED_MODE].sig_out0_idx + config->channel_idx, config->output_invert, 0);
         ^~~~~~~~~~~~~~~
         gpio_iomux_out
../components/lvgl_esp32_drivers/lvgl_tft/esp_lcd_backlight.c:65:9: error: implicit declaration of function 'gpio_pad_select_gpio'; did you mean 'esp_rom_gpio_pad_select_gpio'? [-Werror=implicit-function-declaration]
         gpio_pad_select_gpio(config->gpio_num);
         ^~~~~~~~~~~~~~~~~~~~
         esp_rom_gpio_pad_select_gpio
../components/lvgl_esp32_drivers/lvgl_tft/esp_lcd_backlight.c:67:43: error: 'SIG_GPIO_OUT_IDX' undeclared (first use in this function); did you mean 'GPIO_NUM_MAX'?
         gpio_matrix_out(config->gpio_num, SIG_GPIO_OUT_IDX, config->output_invert, false);
                                           ^~~~~~~~~~~~~~~~
                                           GPIO_NUM_MAX
../components/lvgl_esp32_drivers/lvgl_tft/esp_lcd_backlight.c:67:43: note: each undeclared identifier is reported only once for each function it appears in

../components/lvgl_esp32_drivers/lvgl_helpers.c: In function 'lvgl_interface_init':
../components/lvgl_esp32_drivers/lvgl_helpers.c:107:32: error: 'SPI_DMA_CH1' undeclared (first use in this function); did you mean 'SPI_DMA_CH_AUTO'?
         spi_max_transfer_size, SPI_DMA_CH1,
                                ^~~~~~~~~~~
                                SPI_DMA_CH_AUTO
../components/lvgl_esp32_drivers/lvgl_helpers.c:107:32: note: each undeclared identifier is reported only once for each function it appears in

../components/lvgl_esp32_drivers/lvgl_tft/ili9481.c: In function 'ili9481_init':
../components/lvgl_esp32_drivers/lvgl_tft/ili9481.c:74:5: error: implicit declaration of function 'gpio_pad_select_gpio'; did you mean 'esp_rom_gpio_pad_select_gpio'? [-Werror=implicit-function-declaration]
     gpio_pad_select_gpio(ILI9481_DC);
     ^~~~~~~~~~~~~~~~~~~~
     esp_rom_gpio_pad_select_gpio

../components/lvgl_esp32_drivers/lvgl_tft/ili9486.c: In function 'ili9486_init':
../components/lvgl_esp32_drivers/lvgl_tft/ili9486.c:67:5: error: implicit declaration of function 'gpio_pad_select_gpio'; did you mean 'esp_rom_gpio_pad_select_gpio'? [-Werror=implicit-function-declaration]
     gpio_pad_select_gpio(ILI9486_DC);
     ^~~~~~~~~~~~~~~~~~~~
     esp_rom_gpio_pad_select_gpio

../components/lvgl_esp32_drivers/lvgl_tft/st7735s.c: In function 'st7735s_init':
../components/lvgl_esp32_drivers/lvgl_tft/st7735s.c:102:9: error: implicit declaration of function 'gpio_pad_select_gpio'; did you mean 'esp_rom_gpio_pad_select_gpio'? [-Werror=implicit-function-declaration]
         gpio_pad_select_gpio(ST7735S_DC);
         ^~~~~~~~~~~~~~~~~~~~
         esp_rom_gpio_pad_select_gpio

../components/lvgl_esp32_drivers/lvgl_tft/sh1107.c: In function 'sh1107_init':
../components/lvgl_esp32_drivers/lvgl_tft/sh1107.c:97:9: error: implicit declaration of function 'gpio_pad_select_gpio'; did you mean 'esp_rom_gpio_pad_select_gpio'? [-Werror=implicit-function-declaration]
         gpio_pad_select_gpio(SH1107_DC);
         ^~~~~~~~~~~~~~~~~~~~
         esp_rom_gpio_pad_select_gpio

../components/lvgl_esp32_drivers/lvgl_tft/hx8357.c: In function 'hx8357_init':
../components/lvgl_esp32_drivers/lvgl_tft/hx8357.c:160:5: error: implicit declaration of function 'gpio_pad_select_gpio'; did you mean 'esp_rom_gpio_pad_select_gpio'? [-Werror=implicit-function-declaration]
     gpio_pad_select_gpio(HX8357_DC);
     ^~~~~~~~~~~~~~~~~~~~
     esp_rom_gpio_pad_select_gpio

../components/lvgl_esp32_drivers/lvgl_tft/st7796s.c: In function 'st7796s_init':
../components/lvgl_esp32_drivers/lvgl_tft/st7796s.c:83:2: error: implicit declaration of function 'gpio_pad_select_gpio'; did you mean 'esp_rom_gpio_pad_select_gpio'? [-Werror=implicit-function-declaration]
  gpio_pad_select_gpio(ST7796S_DC);
  ^~~~~~~~~~~~~~~~~~~~
  esp_rom_gpio_pad_select_gpio
```

_Originally posted by @arktrin in https://github.com/lvgl/lvgl_esp32_drivers/issues/164#issuecomment-1009696098_

C47D, 2022-01-11:
It seems to me that those `gpio` functions are in this path https://github.com/espressif/esp-idf/blob/master/components/esp_rom/include/esp32/rom/gpio.h, I will investigate if the proper way to use them is via the suggested functions, such as `esp_rom_gpio_pad_select_gpio`

C47D, 2022-01-11:
`SPI_DMA_CH1` is not defined for ESP32C3 https://github.com/espressif/esp-idf/blob/18af031791608460b65097eb5326aa28e7778d5f/components/driver/include/driver/spi_common.h#L69-L76

arktrin, 2022-01-11:
I've thought this part:
```
#if defined (CONFIG_IDF_TARGET_ESP32C3)
    dma_channel = SPI_DMA_CH_AUTO;
#endif
```
already been added in #153.

C47D, 2022-01-11:
@arktrin Yup, it is, but we currently do the following inside `lvgl_interface_init`. The `SPI_DMA_CH1` symbol is not defined for ESP32C3, so it fails the compilation. The check added in #153 should be done before calling `lvgl_spi_driver_init` instead of inside.

```c
    lvgl_spi_driver_init(TFT_SPI_HOST,
        miso, DISP_SPI_MOSI, DISP_SPI_CLK,
        spi_max_transfer_size, SPI_DMA_CH1,
        DISP_SPI_IO2, DISP_SPI_IO3);
```

C47D, 2022-01-12:
@arktrin I've pushed this branch `fix/cleanup_lvgl_helpers` with a bunch of the errors you pointed out fixed, can you try it?

arktrin, 2022-01-12:
@C47D I don't know what I'm doing wrong but now I can't even configure the `feat/new_driver_test` branch of `lv_port_esp32` project. This is true for the 4.3 release and the latest `master` branch of esp-idf. Also true for latest `fix/cleanup_lvgl_helpers` and `develop` branches of `lvgl_esp32_drivers`.

Here is the [content of CMakeOutput.log](https://pastebin.com/Vt7gMZ9c). Can't figure out which parts are related to errors.

C47D, 2022-01-12:
Thanks @arktrin I will try to add check it later today.

arktrin, 2022-01-12:
@C47D Sorry! I forgot to recursively clone the repository. Now with  the latest `feat/new_driver_test` branch of `lv_port_esp32` and with the  latest `fix/cleanup_lvgl_helpers`  branch of the `lvgl_esp32_drivers` I'm getting single error:
```
../main/main.c: In function 'guiTask':
../main/main.c:85:25: error: 'st7789_flush' undeclared (first use in this function)
     disp_drv.flush_cb = st7789_flush;
                         ^~~~~~~~~~~~
../main/main.c:85:25: note: each undeclared identifier is reported only once for each function it appears in
../main/main.c:94:5: error: implicit declaration of function 'st7789_init' [-Werror=implicit-function-declaration]
     st7789_init(&disp_drv);
     ^~~~~~~~~~~
```

C47D, 2022-01-12:
Damn, I wonder why I don't get those errors, maybe I need to test on a recently cloned repo. Will try to fix it later today, it's just a header include.

C47D, 2022-01-12:
No problem, thanks for the update, I wonder why I don't see that errors in the CI

tore-espressif, 2022-02-03:
Can we close this?

arktrin, 2022-02-03:
Yes, we can.

closed 2022-02-03 by tore-espressif

e5h, 2023-03-22:
Issue still exists when tested on IDF version 5.0