lvgl/lvgl_esp32_drivers #176: Using ESP-LCD component as a display driver design

sukesh-ak, 2022-02-01:
Found that Espressif has a [ESP-LCD component](https://github.com/espressif/esp-idf/tree/master/components/esp_lcd) with documentation but only few [example drivers](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd).
Maybe its a better idea to build/port drivers using this design although not sure the efforts required for it now.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/lcd.html

C47D, 2022-02-02:
Hi,

I think it's being currently being used in the backlight control.

We're currently working (in the devel branch) to abstract the communication interface from the display drivers, with the goal of being able to add new communication interfaces easily.

Any tips @tore-espressif?

sukesh-ak, 2022-02-02:
Somehow I feel everything is messed up with the v8 migration.
Sharing my feedback from another issue on the esp32 port repo [here](https://github.com/lvgl/lv_port_esp32/issues/290)

I tried the following steps

1. Create a fresh ESP-IDF project
2. Add lvgl (from main repo) as an IDF component
3. Add lvgl_esp32_drivers (develop branch) as a IDF component =compiles fine

Now I need an example to test. Could not find one with the latest functions. Every one of the sample was throwing errors because some function names have also changed during the port and probably some issues with kconfig too.

Do you have a working ESP-IDF+v8 sample which shows just a label?

When you enable touch, the I2C port settings won't show up, when you go back to menuconfig, there is a new option to enable and define the I2C pins.

I am using Wireless-Tag WT32-SC01 3.5" 480x320 board with ST7796s and FT6336U.

Tried v7 using the master and it works but extremely slow.
Tested the display using Arduino+TFT_eSPI and performs great.

sukesh-ak, 2022-02-02:
Here is my sample main.c
What am I missing? Nothing shows up in the display.

C47D, 2022-02-02:
The portv8 branch of that repo is several months old, it was a leftover branch I used to test LVGL before v8 was officially released. I forgot to delete it and I've seen some people using it. 

We got an example project on the CI of this repo using LVGL v8, having support for both versions in a single main file is a bit harder and uglier, but I can work on it only on my free time.

sukesh-ak, 2022-02-02:
> The portv8 branch of that repo is several months old, it was a leftover branch I used to test LVGL before v8 was officially released. I forgot to delete it and I've seen some people using it.
>
> We [...]

I didn't use the v8 port but pulled from the main lvgl rep.
I don't need v7. I need v8 with IDF working sample :)

sukesh-ak, 2022-02-02:
And here is the difference in out between different versions

[image]

sukesh-ak, 2022-02-02:
So I copied main.c from v7 to v8, and trying to make the required changes.

v8
lvgl_driver_init() function is missing. I believe its replaced with lvgl_interface_init()

lvgl_helpers.c
lvgl_get_display_buffer_size() returns 0 because for LVGL 8 there is a TODO statement - so not implemented.

v8 API changes?
changed 'lv_disp_buf_t' to 'lv_disp_draw_buf_t'
changed 'disp_drv.buffer' to 'disp_drv.draw_buf'

tore-espressif, 2022-02-03:
Hello @sukesh-ak 
If you really need LVGL v8, then the `esp_lcd` component, that you mentioned, is a better place to start, as the examples for `esp_lcd` were written exclusively for v8.
ST7796s and ST7789 have identical interface, so you can start with this example: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/i80_controller
and modify it for SPI (I can help with that).

sukesh-ak, 2022-02-03:
> Hello @sukesh-ak If you really need LVGL v8, then the `esp_lcd` component, that you mentioned, is a better place to start, as the examples for `esp_lcd` were written exclusively for v8. ST7796s and [...]

I have been a software guy all life, so have not gone into LCD driver side of things. So its a big of grey area for me.

> Initialize Intel 8080 bus

What is this portion of the code for? Is it using parellel interface instead of SPI?

[Is this](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/tjpgd) 'examples/peripherals/lcd/tjpgd' a better sample to refer?

> I can help with that

Thank you.

sukesh-ak, 2022-02-03:
@tore-espressif Appreciate your help.

Wireless Tag WT32-SC01

> 3.5" 480x320 ST7796S TFT Display
Capacitive touchscreen FT6336U

    - TFT_RST=22
    - TFT_SCLK=14
    - TFT_DC=21
    - TFT_CS=15
    - TFT_MOSI=13
    - TFT_MISO=-1
    - TFT_BCKL=23

    - TOUCH_DRIVER=0x6336
    - TOUCH_SDA=18
    - TOUCH_SCL=19
	
    - TOUCH_IRQ=-1   ; not connected
    - TOUCH_RST=-1   ; not used, connected to 3.3V
    - I2C_TOUCH_FREQUENCY=400000
    - I2C_TOUCH_ADDRESS=0x38

Datasheet - ST7796s
https://www.displayfuture.com/Display/datasheet/controller/ST7796s.pdf

tore-espressif, 2022-02-07:
Sorry for the late reply, I got caught with other issues.

>  Is it using parellel interface instead of SPI?

Yes, Intel 8080 interface is parallel interface. You need to change it to SPI.

> [Is this](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/tjpgd) 'examples/peripherals/lcd/tjpgd' a better sample to refer?

This example uses SPI interface but misses the LVGL porting part. It is a good starting point to verify that your setup is correct.

> Also what about capacitive touch (I2C) while using esp_lcd ?

Touch controller is not part of `esp_lcd` and we don't have similar component for touch controllers yet. However, touch controllers are a lot simpler to handle than display controllers, so you should be able to copy-paste it from this repo

sukesh-ak, 2022-02-07:
@tore-espressif 

What about these libraries? Looks like a lot of code for relevant displays are already written.
The list of supported Display drivers looks complete and the sample uses lvgl too.

https://github.com/espressif/esp-dev-kits/blob/9f7a9b56f28f399b5bf383949c26d900375adedd/esp32-s2-hmi-devkit-1/components/screen/CMakeLists.txt#L3

sukesh-ak, 2022-02-22:
Finally found LovyanGFX the best drivers which works for both Arduino and ESP-IDF.

ESP-IDF & Arduino samples for WT32-SC01 using LVGL8 + LOVYANGFX 

Device : WT32-SC01 
Display / Touch Drivers : LovyanGFX
Widgets : LVGL 8.x

ESP-IDF => WT32-SC01 + LVGL8 + LOVYANGFX 
https://github.com/sukesh-ak/LVGL8-WT32-SC01-IDF

Arduino => WT32-SC01 + LVGL8 + LOVYANGFX 
https://github.com/sukesh-ak/LVGL8-WT32-SC01-Arduino