lvgl/lvgl_esp32_drivers #178: ST7789 SPI mode seems wrong

richardclli, 2022-02-06:
I have a display that is not working with the driver in SPI mode 2, but when changing the mode to 0, it works.

Not sure why the following code is needed in the first place.

#if defined (CONFIG_LV_TFT_DISPLAY_CONTROLLER_ST7789)
#define SPI_TFT_SPI_MODE    (2)
#else
#define SPI_TFT_SPI_MODE    (0)
#endif

richardclli, 2022-02-06:
After checking with eSPI_TFT library, it seems some ST7789 uses mode 0 and some uses mode 2/3

C47D, 2022-02-06:
I've been using it in mode 2 for quite a while now, and it's working fine. I'll try to add an option in the Kconfig menu to choose the mode. Thanks for the report.

richardclli, 2022-02-07:
A good choice, have a config is better, thanks.  I am using the version in the esp32 port, which is quite old, but I checked newer versions still use mode 2 hardcoded.

richardclli, 2022-02-07:
After tryout, I am creating a project using lvgl and the esp32 driver as submodules, may try newer version.  Use master branch in esp32 drivers is good?  Or should I move to another branch?  Which version of LVGL should I use?

C47D, 2022-02-07:
That project only supports LVGL v7, we're trying to get a general cleanup on the repo before starting to support LVGL v8. I will send you the patch later today.

richardclli, 2022-02-08:
Take your time, I already have a quick fix anyway.

ksmq, 2022-04-12:
TTGO T-Watch which uses the ST7789 seems to work with both spi mode 0 and 3, but not with 2. I've seen other libs also use 0. Took me a while to debug this so posting in case someone else comes across the same problem.

PR #181 (2022-02-15): Allow user to configure custom SPI mode
Fixes #178
commit 7c6bdc56 (2022-02-15): Allow user to configure custom SPI mode (#181)

Fixes #178