lvgl/lvgl_esp32_drivers #190: Upgrade support for ESP32-S3 series

Huckies, 2022-04-09:
A esp32s3 is more like a esp32c3 which only supports ```SPI_DMA_CH_AUTO```, so it might be better to upgrade `lvgl_esp32_drivers/lvgl_helpers.c` line 178

Also I met an assert error compiling (the target chip is esp32s3) and solved it by adding ```"-DSPI_HOST_MAX=3"```to line 91 of lvgl_esp32_drivers/CMakeLists.txt```

tigerya214, 2022-04-16:
Hi, you answer hit me!!! do you know why it show "'TFT_SPI_HOST' undeclared (first use in this function); did you mean 'SPI3_HOST'?" when I am compiling the lvgl?

Huckies, 2022-04-16:
Maybe you want to config it before compiling, or add a declaration on your own header file.

tigerya214, 2022-04-16:
[image]
就是编译后出现这个问题，是不是LVGL的函数因为版本原因变了？
（看了下好像是国人，哈哈！）