lvgl/lvgl_esp32_drivers #193: Compilation errors in esp_helpers.c

tigerya214, 2022-04-17:
I have two problem about compile the esp_helpers.c, 
1、I think it is not support the esp32-s3, because I find there are not esp32-s3 in line 200 to  /* Initialize spi bus master */
2、what compiler should I choose? (I am not using the windows-msvc-x64)

`In file included from ../components/lvgl_esp32_drivers/lvgl_helpers.h:17,
                 from ../components/lvgl_esp32_drivers/lvgl_helpers.c:10:
../components/lvgl_esp32_drivers/lvgl_helpers.c: In function 'lvgl_driver_init':
../components/lvgl_esp32_drivers/lvgl_spi_conf.h:68:22: error: 'HSPI_HOST' undeclared (first 
use in this function); did you mean 'SPI3_HOST'?
 #define TFT_SPI_HOST HSPI_HOST
                      ^~~~~~~~~
../components/lvgl_esp32_drivers/lvgl_helpers.c:111:26: note: in expansion of macro 'TFT_SPI_HOST'
     lvgl_spi_driver_init(TFT_SPI_HOST,
                          ^~~~~~~~~~~~
../components/lvgl_esp32_drivers/lvgl_spi_conf.h:68:22: note: each undeclared identifier is reported only once for each function it appears in
 #define TFT_SPI_HOST HSPI_HOST
                      ^~~~~~~~~
../components/lvgl_esp32_drivers/lvgl_helpers.c:111:26: note: in expansion of macro 'TFT_SPI_HOST'
     lvgl_spi_driver_init(TFT_SPI_HOST,
                          ^~~~~~~~~~~~
In file included from d:\software\esp-idf\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\sys\reent.h:503,
                 from D:/software/ESP-IDF/esp-idf/components/newlib/platform_include/sys/reent.h:17,
                 from d:\software\esp-idf\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\sys-include\string.h:11,
                 from ../components/lvgl/src/lv_hal/../lv_misc/lv_area.h:17,
                 from ../components/lvgl/src/lv_hal/lv_hal_disp.h:22,
                 from ../components/lvgl/src/lv_hal/lv_hal.h:16,
                 from ../components/lvgl/lvgl.h:30,
                 from ../components/lvgl_esp32_drivers/lvgl_tft/disp_driver.h:16,
                 from ../components/lvgl_esp32_drivers/lvgl_helpers.h:18,
                 from ../components/lvgl_esp32_drivers/lvgl_helpers.c:10:
../components/lvgl_esp32_drivers/lvgl_helpers.c: In function 'lvgl_spi_driver_init':
../components/lvgl_esp32_drivers/lvgl_helpers.c:218:13: error: 'SPI_HOST' undeclared (first use in this function); did you mean 'SPI3_HOST'?
     assert((SPI_HOST <= host) && (HSPI_HOST >= host));
             ^~~~~~~~
../components/lvgl_esp32_drivers/lvgl_helpers.c:218:35: error: 'HSPI_HOST' undeclared (first 
use in this function); did you mean 'SPI3_HOST'?
     assert((SPI_HOST <= host) && (HSPI_HOST >= host));
                                   ^~~~~~~~~
ninja: build stopped: subcommand failed.
终端进程“C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja ”已终止，退出代码: 1。
`

Silvesterrr, 2022-06-02:
Same problem

paranormal-programmer, 2023-04-13:
ESP32 and ESP32-S3 have different SPI names for setup. HSPI and VSPI is for ESP32. 
 Aside from that, in my experience there are several compatibility issues with current lvgl and lvgl_esp32_drivers because the drivers haven't received support in a while. But this post has helped me: https://www.esp32.com/viewtopic.php?f=40&t=32455&p=110662#p110662
 Or use lvgl 7.11