lvgl/lvgl_esp32_drivers #202: SPI_HOST_MAX, LV_VER_RES_MAX and LV_HOR_RES_MAX not defined!

mohammadjavadpirhadi, 2022-07-26:
Hello,
I'm using lvgl v8.3.1 and the master branch of lvgl_esp32_drivers, but during the build, I get these errors:

../components/lvgl_esp32_drivers/lvgl_helpers.c:157:28: error: 'SPI_HOST_MAX' undeclared (first use in this function);
../components/lvgl_esp32_drivers/lvgl_helpers.c:157:28: error: 'LV_VER_RES_MAX' undeclared (first use in this function);
../components/lvgl_esp32_drivers/lvgl_helpers.c:157:28: error: 'LV_HOR_RES_MAX' undeclared (first use in this function);

xbzxsh, 2022-07-29:
add these lines under the define comment in file "lvgl_helpers.h" should work.

# define LV_HOR_RES_MAX 320
# define LV_VER_RES_MAX 240
# define SPI_HOST_MAX 3

mohammadjavadpirhadi, 2022-07-29:
> add these lines under the define comment in file "lvgl_helpers.h" should work.
> 
> ```c
> # define LV_HOR_RES_MAX 320
> # define LV_VER_RES_MAX 240
> # define SPI_HOST_MAX 3
> ```

@xbzxsh Yes, I know! But I expect that the library handle this automatically.

MrSurly, 2022-12-05:
Was there any resolution to this?  In the [other issue](https://github.com/lvgl/lvgl/issues/3521) related to this, @kisvegabor said:

> It's because the ESP related repos are still not updated to LVGL v8.

So I reverted `components/lvgl` in my project to `v7.9.1`.

Also, I'm using IDF `v4.4.3` since the [LVGL ESP32 docs](https://docs.lvgl.io/latest/en/html/get-started/espressif.html#get-the-lvgl-demo-project-for-esp32) say 

> ESP-IDF v4 framework is the suggested version to use.

Did anyone find the magic combination to make LVGL work with ESP32?

filgra96, 2023-04-13:
I saw this issue isn't closed yet, so I'm goint to write what I tried.

I'm a newcomer in LVGL and I'd like to develop a little project with the ESP32. However **I noticed from the docs that It's not clear for which version is made the `lvgl_esp32_drivers`**.

I'm using the `v4.4.1` version for ESP-IDF.

I created a very simple piece of code:
```c
#include "lvgl.h"

int app_main(void)
{
    lv_init();

    return 0;
}
```

Directory for `lvgl` library under `components/lvgl`.
Directory for `lvgl_esp32_drivers` under `components/lvgl_esp32_drivers` at `master` branch.

I started with the master branch for `lvgl` and I found this error:
```
../components/lvgl/src/core/../libs/ffmpeg/../../../src/widgets/meter/lv_meter.h:22:2: error: #error "lv_meter: Complex drawing is required. Enable it in lv_conf.h (LV_USE_DRAW_MASKS 1)"
 #error "lv_meter: Complex drawing is required. Enable it in lv_conf.h (LV_USE_DRAW_MASKS 1)"
  ^~~~~
In file included from ../components/lvgl/src/core/../libs/ffmpeg/lv_ffmpeg.h:15,
                 from ../components/lvgl/src/core/lv_obj.c:28:
../components/lvgl/src/core/../libs/ffmpeg/../../../lvgl.h:120:2: warning: #warning "You are using the development version of LVGL which is not stable at this moment. For production use the release/v8.3 branch. To silence this warning add #define LV_USE_DEV_VERSION to lv_conf.h" [-Wcpp]
 #warning "You are using the development version of LVGL which is not stable at this moment. For production use the release/v8.3 branch. To silence this warning add #define LV_USE_DEV_VERSION to lv_conf.h"
  ^~~~~~~
```

I switched to `release/v8.3`, here the resulting error, as a lot of people said:
```
../components/lvgl_esp32_drivers/lvgl_helpers.c: In function 'lvgl_driver_init':
../components/lvgl_esp32_drivers/lvgl_helpers.h:57:25: error: 'LV_HOR_RES_MAX' undeclared (first use in this function); did you mean 'LV_HOR_RES'?
 #define DISP_BUF_SIZE  (LV_HOR_RES_MAX * 40)
                         ^~~~~~~~~~~~~~
C:/Espressif/frameworks/esp-idf-v4.4.1/components/log/include/esp_log.h:421:137: note: in expansion of macro 'DISP_BUF_SIZE'
         if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR,      tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                                                                         ^~~~~~~~~~~
C:/Espressif/frameworks/esp-idf-v4.4.1/components/log/include/esp_log.h:443:41: note: in expansion of macro 'ESP_LOG_LEVEL'
         if ( LOG_LOCAL_LEVEL >= level ) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
                                         ^~~~~~~~~~~~~
C:/Espressif/frameworks/esp-idf-v4.4.1/components/log/include/esp_log.h:353:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
                                      ^~~~~~~~~~~~~~~~~~~
../components/lvgl_esp32_drivers/lvgl_helpers.c:61:5: note: in expansion of macro 'ESP_LOGI'
     ESP_LOGI(TAG, "Display buffer size: %d", DISP_BUF_SIZE);
     ^~~~~~~~
../components/lvgl_esp32_drivers/lvgl_helpers.h:57:25: note: each undeclared identifier is reported only once for each function it appears in
 #define DISP_BUF_SIZE  (LV_HOR_RES_MAX * 40)
                         ^~~~~~~~~~~~~~
C:/Espressif/frameworks/esp-idf-v4.4.1/components/log/include/esp_log.h:421:137: note: in expansion of macro 'DISP_BUF_SIZE'
         if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR,      tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                                                                         ^~~~~~~~~~~
C:/Espressif/frameworks/esp-idf-v4.4.1/components/log/include/esp_log.h:443:41: note: in expansion of macro 'ESP_LOG_LEVEL'
         if ( LOG_LOCAL_LEVEL >= level ) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
                                         ^~~~~~~~~~~~~
C:/Espressif/frameworks/esp-idf-v4.4.1/components/log/include/esp_log.h:353:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
                                      ^~~~~~~~~~~~~~~~~~~
../components/lvgl_esp32_drivers/lvgl_helpers.c:61:5: note: in expansion of macro 'ESP_LOGI'
     ESP_LOGI(TAG, "Display buffer size: %d", DISP_BUF_SIZE);
     ^~~~~~~~
In file included from c:\espressif\tools\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\sys-include\sys\reent.h:503,
                 from C:/Espressif/frameworks/esp-idf-v4.4.1/components/newlib/platform_include/sys/reent.h:17,
                 from c:\espressif\tools\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\sys-include\string.h:11,
                 from ../components/lvgl/src/misc/lv_mem.h:20,
[... 18 lines cut ...]
                 from ../components/lvgl_esp32_drivers/lvgl_tft/disp_driver.h:16,
                 from ../components/lvgl_esp32_drivers/lvgl_helpers.h:18,
                 from ../components/lvgl_esp32_drivers/lvgl_helpers.c:10:
../components/lvgl_esp32_drivers/lvgl_helpers.c: In function 'lvgl_spi_driver_init':
../components/lvgl_esp32_drivers/lvgl_helpers.c:157:28: error: 'SPI_HOST_MAX' undeclared (first use in this function); did you mean 'GPIO_PORT_MAX'?
[... cut ...] pi_driver_init()` inside `lvgl_helpers.c`. Infact it doesn't use the last `master` for `lvgl_esp32_drivers`, but the commit `9fed1cc47b5a45fec6bae08b55d2147d3b50260c` instead.

> Did anyone find the magic combination to make LVGL work with ESP32?

Related to this question: I didn't find a clear way to compile, as you can see.

---

I hope I shared useful elements to improve this library. **Only a question remains: using the missing definitions, can I use the LVGL library without bad surprises?**