lvgl/lvgl_esp32_drivers #220: gpio_matrix_out is not defined

cjheath, 2023-03-21:
https://github.com/lvgl/lvgl_esp32_drivers/blob/26fe6e7703162ac9b0ee82405e858dc37b885d1f/lvgl_tft/esp_lcd_backlight.c#L59-L67

The called function `gpio_matrix_out` does not seem to exist anywhere in this library, in LVGL, or in any version of ESP-IDF.

What is the intention of this code, and how does anyone manage to build this package without a definition of this function?
More importantly, can it be fixed please?

e5h, 2023-03-22:
I am having the same issue, I found some discussion here:

https://github.com/lvgl/lvgl_esp32_drivers/pull/171

Not sure how this is not fixed. I am going to try to swap the functions to the esp_rom ones instead.

cjheath, 2023-03-22:
That's not going to work for me because neither ESP-IDF v4.4.4 nor the current trunk version contains any reference to `esp_rom_gpio_connect_out_signal`

What version of ESP-IDF are you using?

e5h, 2023-03-22:
I am using v5.0, there is a reference to it [here](https://github.com/espressif/esp-idf/blob/master/components/esp_rom/include/esp_rom_gpio.h), the include statement in my first comment worked for me on an esp32-c3

On a side note, it would have been nice if the devs put some warnings about incompatible versions. As it stands right now, this library seems partially incompatible with both v8 of LVGL and v5.0 of the ESP-IDF.

cjheath, 2023-03-22:
Oh, ok, I missed that - I was probably searching fir any definition of `gpio_matrix_out` at the time. I didn't want to jump to v5 yet because I figured that 3rd party support for V4 would be more stable.

I think I'll be able to figure out something that works with v4. It was just not knowing the definition for `gpio_matrix_out` I didn't know if it might contain any subtleties that are necessary to this function.

Pi-pythoner, 2023-04-23:
Could anybody share your codes after edited with the new GPIO?

hiruna, 2023-05-26:
feel free to try this branch out https://github.com/hiruna/lvgl_esp32_drivers/tree/develop/lvgl_8.3.7_idf_5.2