lvgl/lvgl_esp32_drivers #222: LVGL 8.3.7 & ESP-IDF 5.2

hiruna, 2023-05-26:
Hi all,

I got IDF v5.2 (master) & [LVGL 8.3.7](https://github.com/lvgl/lvgl/commit/2b56e04205481daa6575bd5f7ab5df59d11676eb) to compile.

I only have a SSD1306 128x64 OLED display and a 128x128 ST7735 LCD but focusing on the SSD1306 OLED for now.

Will add a branch for the `lv_port_esp32` soon, the main.c needs some tweaking (Only included changes for the SSD1306 hello world, `lv_demo_*` functions will need changes too):

`static lv_disp_buf_t disp_buf;` -> `static lv_disp_draw_buf_t disp_buf;`

`lv_disp_buf_init(&disp_buf, buf1, buf2, size_in_px);` -> `lv_disp_draw_buf_init(&disp_buf, buf1, buf2, size_in_px);`

`disp_drv.buffer = &disp_buf;` -> `disp_drv.draw_buf = &disp_buf;`

`lv_obj_t * label1 =  lv_label_create(scr, NULL);` -> `lv_obj_t * label1 =  lv_label_create(scr);`

`lv_obj_align(label1, NULL, LV_ALIGN_CENTER, 0, 0);` -> `lv_obj_align(label1, LV_ALIGN_CENTER, 0, 0);`

Compilation & flashing succeeds but my OLED display shows garabge (pixel noise), I'm thinking something's wrong with the i2c_manager ?

Feel free to test this branch out and let me know how it is.

hiruna, 2023-05-27:
Ok.. so I got my SSD1306 OLED display working again...

See https://github.com/hiruna/lv_port_esp32/tree/develop/lvgl_8.3.7_idf_5.2 for the `lv_port_esp32` 

Added https://github.com/hiruna/lv_port_esp32/blob/develop/lvgl_8.3.7_idf_5.2/sdkconfig.ssd1306 for reference 

And https://github.com/hiruna/lvgl_esp32_drivers/tree/develop/lvgl_8.3.7_idf_5.2

hiruna, 2023-05-27:
128x128 ST7735 LCD is somewhat working... trying to figure out why the rendering is off the screen bounds 🤔 
[image]

hiruna, 2023-05-27:
Fixed it in https://github.com/hiruna/lvgl_esp32_drivers/blob/develop/lvgl_8.3.7_idf_5.2/lvgl_tft/st7735s.h
[image]

hiruna, 2023-05-27:
Tested lvgl benchmark, 128x128 display is obviously too small but good to see it working 
[image]

hiruna, 2023-05-28:
Got my hands on a ILI9341 3.2" 320x240 display w/ touch. Touch panel was not connected/tested in this test.

ESP32 CPU Speed: 240MHz
SPI Speed: 80MHz

Pins:

```CS -> GPIO15
CLK -> GPIO14
MISO -> GPIO12
MOSI -> GPIO13
LED -> GPIO26
DC -> GPIO27
RST -> GPIO33
```

sdkconfig used: https://github.com/hiruna/lv_port_esp32/blob/develop/lvgl_8.3.7_idf_5.2/sdkconfig.ili9341

[image]

hiruna, 2023-05-28:
My ILI9341 3.2" 320x240 touch panel has a HR2046 instead of a XPT2046 but I configured it as a XPT2046 and it works for me. 

```
CONFIG_LV_TOUCH_CONTROLLER_SPI3_HOST=y
CONFIG_LV_TOUCH_X_MIN=200
CONFIG_LV_TOUCH_Y_MIN=200
CONFIG_LV_TOUCH_X_MAX=1907
CONFIG_LV_TOUCH_Y_MAX=1887
CONFIG_LV_TOUCH_SPI_MISO=19
CONFIG_LV_TOUCH_SPI_MOSI=23
CONFIG_LV_TOUCH_SPI_CLK=18
CONFIG_LV_TOUCH_SPI_CS=5
CONFIG_LV_TOUCH_PIN_IRQ=25
CONFIG_LV_TOUCH_INVERT_X=y
CONFIG_LV_TOUCH_DETECT_IRQ=y
```

hiruna, 2023-05-28:
Tested with [Waveshare GC9A01 240x240 round display module](https://www.waveshare.com/1.28inch-lcd-module.htm)

sdkconfig: https://github.com/hiruna/lv_port_esp32/blob/develop/lvgl_8.3.7_idf_5.2/sdkconfig.GC9A01

[image]

[image]

[image]

hiruna, 2023-05-30:
got my self a Freenove esp32s3 (N8R8) and a 1.9inch 170×320 LCD Display Module (ST7789V2) (https://www.waveshare.com/1.9inch-lcd-module.htm)

262K colors are supported by display but LVGL doesnt have the option
took some trial and error to setup the PSRAM for esp32s3 
sdkconfig: https://github.com/hiruna/lv_port_esp32/blob/develop/lvgl_8.3.7_idf_5.2/sdkconfig.esp32s3_ST7789

also had to set spi dma chan to auto (got an error saying chan 1 was not supported by the chip, maybe this is not for everyone ?) - https://github.com/hiruna/lvgl_esp32_drivers/commit/f22026caf7e751575a17ed3f268621ad90fc205e

added offsets for the 170x320 display - https://github.com/hiruna/lvgl_esp32_drivers/commit/16f9f93dab43f492fb444df42c61d11da790fdff

[image]

[image]

[image]

[image]

hiruna, 2023-06-01:
Ported https://github.com/jakpaul/lvgl_touch_calibration via my fork at https://github.com/hiruna/lvgl_touch_calibration.

[PR](https://github.com/jakpaul/lvgl_touch_calibration/pull/3) submitted to @jakpaul with my changes.

Updated https://github.com/hiruna/lv_port_esp32/blob/develop/lvgl_8.3.7_idf_5.2/main/main.c to cater for the touch calibration.
* If the a touch screen is configured, the demo will first begin the touch calibration process (check non volatile storage for existing data, if not calibrate)

Demonstration:

https://github.com/lvgl/lvgl_esp32_drivers/assets/3973928/d169e1f1-ba7c-43e8-ada5-a7a67f4ec9e5

hiruna, 2023-06-03:
Tested 320x480px [3.5 inch RPi Display ](http://www.lcdwiki.com/3.5inch_RPi_Display) (ILI9486 + XPT2046)
[image]

sdkconfig: https://github.com/hiruna/lv_port_esp32/blob/develop/lvgl_8.3.7_idf_5.2/sdkconfig.ILI9486_rpi_disp

hiruna, 2023-06-03:
added [backward compatibility with esp-idf release/v4.4](https://github.com/hiruna/lvgl_esp32_drivers/commit/de52f295db7e91165166eb4c9d8bf9aba43cb239)

hiruna, 2023-06-03:
added [backward compatibility with esp-idf release/v4.2 (<4.3)](https://github.com/hiruna/lvgl_esp32_drivers/commit/a76c88fd66cf611a5ace4875c3791c778218f29b)

hiruna, 2023-06-03:
Tested ESP-IDF versions:

* master
* release/v5.1
* release/v5.0
* release/v4.4
* release/v4.3
* release/v4.2

mhaberler, 2023-07-17:
Hi Hiruna

just a status report

I tried this branch grafted onto https://github.com/mhaberler/m5core2_esp-idf_demo.git , modified from [ropg](https://github.com/ropg/m5core2_esp-idf_demo)

this works fine except for the touch controller, which I still have to look into - looks like the FT6X36 is found and properly initialized (see below), but needs some love to actually report ;)

[image]

I'm primarily interested in M5Stack Core2 and CoreS3 so dabbling with those

great job!

- Michael

ps: complete boot log:
`````
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7132
ho 0 tail 12 room 4
load:0x40078000,len:15740
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3884
entry 0x40080650
I (33) boot: ESP-IDF v5.2-dev-1709-g4fc2e5cb95-dirty 2nd stage bootloader
I (33) boot: compile time Jul 17 2023 12:24:17
I (35) boot: Multicore bootloader
I (39) boot: chip revision: v3.0
I (43) boot.esp32: SPI Speed      : 80MHz
I (48) boot.esp32: SPI Mode       : DIO
I (52) boot.esp32: SPI Flash Size : 16MB
I (57) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (81) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  2 factory          factory app      00 00 00010000 00100000
I (96) boot: End of partition table
I (100) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=10ef8h ( 69368) map
I (129) esp_image: segment 1: paddr=00020f20 vaddr=3ffb0000 size=02280h (  8832) load
I (133) esp_image: segment 2: paddr=000231a8 vaddr=40080000 size=0ce70h ( 52848) load
I (154) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=380a4h (229540) map
I (223) esp_image: segment 4: paddr=000680cc vaddr=4008ce70 size=03558h ( 13656) load
I (236) boot: Loaded app from partition at offset 0x10000
I (237) boot: Disabling RNG early entropy source...
I (248) cpu_start: Multicore app
I (249) quad_psram: This chip is ESP32-D0WD
I (249) esp_psram: Found 8MB PSRAM device
I (253) esp_psram: Speed: 40MHz
I (256) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (264) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (273) cpu_start: Pro cpu up.
I (277) cpu_start: Starting app cpu, entry point is 0x400813a8
I (0) cpu_start: App cpu up.
I (1188) esp_psram: SPI SRAM memory test OK
I (1195) cpu_start: Pro cpu start user code
I (1196) cpu_start: cpu freq: 240000000 Hz
I (1196) cpu_start: Application information:
I (1199) cpu_start: Project name:     basic
I (1204) cpu_start: App version:      f20f7de-dirty
I (1209) cpu_start: Compile time:     Jul 17 2023 12:24:14
I (1215) cpu_start: ELF file SHA256:  444f948e3...
I (1221) cpu_start: ESP-IDF:          v5.2-dev-1709-g4fc2e5cb95-dirty
I (1228) cpu_start: Min chip rev:     v3.0
I (1233) cpu_start: Max chip rev:     v3.99 
I (1238) cpu_start: Chip rev:         v3.0
I (1243) heap_init: Initializing. RAM available for dynamic allocation:
I (1250) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1256) heap_init: At 3FFC79B0 len 00018650 (97 KiB): DRAM
I (1262) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1269) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1275) heap_init: At 400903C8 len 0000FC38 (63 KiB): IRAM
I (1282) esp_psram: Adding pool of 4096K of PSRAM memory to heap allocator
I (1290) spi_flash: detected chip: generic
I (1294) spi_flash: flash io: dio
I (1298) app_start: Starting scheduler on CPU0
I (1303) app_start: Starting scheduler on CPU1
I (1303) main_task: Started on CPU0
I (1313) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1313) main_task: Calling app_main()
Hello world!
This is esp32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 300, 16MB external flash
Free heap: 4374488
I (1333) m5core2_axp192: Initialising
I (1333) i2c_manager: Starting I2C master at port 0.
I (1343) i2c_manager: Initialised port 0 (SDA: 21, SCL: 22, speed: 1000000 Hz.)
I (1353) m5core2_axp192: 	Vbus limit off
I (1353) m5core2_axp192: 	Speaker amplifier off
I (1363) m5core2_axp192: 	RTC battery charging enabled (3v, 200uA)
I (1363) m5core2_axp192: 	ESP32 power voltage set to 3.35v
I (1373) m5core2_axp192: 	LCD backlight voltage set to 3.20v
I (1383) m5core2_axp192: 	LCD logic and sdcard voltage set to 3.3v
I (1383) m5core2_axp192: 	Vibrator voltage preset to 2v
I (1393) m5core2_axp192: 	LED on
I (1393) m5core2_axp192: 	Charge current set to 100 mA
I (1403) m5core2_axp192: 	Battery voltage now: 4.12 volts
I (1403) m5core2_axp192: 	Power key set, 4 seconds for hard shutdown
I (1413) m5core2_axp192: 	Enabled all ADC channels
I (1423) m5core2_axp192: 	USB / battery powered, 5V bus on
I (1523) m5core2_axp192: 	LCD and touch reset
I (1623) lvgl_i2c: Now following I2C Manager for locking
I (1623) lvgl_helpers: Display hor size: 320, ver size: 240
I (1623) lvgl_helpers: Display buffer size: 12800
I (1623) lvgl_helpers: Initializing SPI master for display
I (1633) lvgl_helpers: Configuring SPI host SPI2_HOST
I (1643) lvgl_helpers: MISO pin: 38, MOSI pin: 23, SCLK pin: 18, IO2/WP pin: -1, IO3/HD pin: -1
I (1653) lvgl_helpers: Max transfer size: 25600 (bytes)
I (1653) lvgl_helpers: Initializing SPI bus...
I (1663) disp_spi: Adding SPI device
I (1663) disp_spi: Clock speed: 40000000Hz, mode: 0, CS pin: 5
I (1673) ILI9341: Initialization.
I (1873) ILI9341: Display orientation: PORTRAIT
I (1873) ILI9341: 0x36 command value: 0x08
I (1873) FT6X36: Found touch panel controller
I (1873) FT6X36: 	Device ID: 0x00
I (1873) FT6X36: 	Chip ID: 0x64
I (1883) FT6X36: 	Device mode: 0x01
I (1883) FT6X36: 	Firmware ID: 0x00
I (1893) FT6X36: 	Release code: 0x00
Running...
`````

hiruna, 2023-07-21:
Hi @mhaberler , 

I don't own a M5Stack/(display with FT6X36 touch controller) but if you enable verbose logging via the sdkconfig (since I set touch point reporting to verbose only via https://github.com/lvgl/lvgl_esp32_drivers/pull/223/commits/50b212aa560dc9f92e38a0d4ef4b62e977e630b6) [it improves performance when it's not always outputting logs per touch]

you can set the default log level to debug via the `CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y` sdkconfig flag

zuckschwerdt, 2023-10-02:
@hiruna thanks for this port. I have a ESP32-2432S032C board and it basically works.

There was one major blocker: it seems that `gpio_iomux_out()` disconnects the LEDC setup.
In [esp_lcd_backlight.c](https://github.com/hiruna/lvgl_esp32_drivers/blob/develop/lvgl_8.3.7_idf_5.2/lvgl_tft/esp_lcd_backlight.c#L63) I had to move that call before the `ledc_` setup (i.e. up two lines). It would also work without that call. Maybe you can test if that change works with your boards and apply that.

Also the display orientation seems mirrored and inverted. I'm not sure I got the right fix but changing the [`ili9341_set_orientation.c`](https://github.com/hiruna/lvgl_esp32_drivers/blob/develop/lvgl_8.3.7_idf_5.2/lvgl_tft/ili9341.c#L216) data to `{0xC0, 0x00, 0xA0, 0x60}` is visually correct for all four orientations.

OscarMonedero, 2024-04-17:
is this compatible with esp32c3?

zixuan945, 2024-11-13:
> is this compatible with esp32c3?

this fork compatible with esp32c3 https://github.com/BunnySakura/lv_port_esp32c3/tree/lvgl_8.3.7_idf_5.2