lvgl/lvgl_esp32_drivers #226: Wrong SPI Mode when using ST7789

Volvox0815, 2023-09-04:
Hello,

the ST7789 driver has a valid signal on SDA the rising edge of the clock. (Datasheet page 34 of the ST7789 driver)

 So in the lvgl_spi_conf.h there is an issue:

```
#if defined (CONFIG_LV_TFT_DISPLAY_CONTROLLER_ST7789)
#define SPI_TFT_SPI_MODE    (2)
#else
#define SPI_TFT_SPI_MODE    (0)

```

The SPI_TFT_SPI_MODE needs to be 3 according to the ESP driver definition

```
 uint8_t mode;                   /**< SPI mode, representing a pair of (CPOL, CPHA) configuration:
                                         - 0: (0, 0)
                                         - 1: (0, 1)
                                         - 2: (1, 0)
                                         - 3: (1, 1)

```
It took me hours to find out what is the issue, because under some reasons a few displays are working even though it is the wrong SPI mode.
Thanks for this great ESP port of the library!

jaylikesbunda, 2025-08-12:
big up this

tototo31, 2025-08-12:
Can confirm as well after weeks of troubleshooting